kbingham/libcamera #17: A fresh install needs to run ldconfig manually

kbingham, 2020-11-20:
It would seem that installing libcamera from clean requires a manual invocation of ldconfig to update the ld paths.
I'd expect this to be handled by the install phase, so perhaps something is missing.

headkaze, 2021-11-30:
I was getting the following error when running `cam --list`:
`cam: error while loading shared libraries: libcamera-base.so.0: cannot open shared object file: No such file or directory`

I then ran `ldconfig` which gave me an error:
`/sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied`

So I ran `sudo ldconfig` then re-compiled and installed libcamera and this time `cam --list` ran as expected.

kbingham, 2021-12-03:
This should all be resolved by:

https://github.com/mesonbuild/meson/issues/4737

closed 2021-12-03 by kbingham

mfreeborn, 2023-02-16:
Just to say, I stumbled across this issue today on my desktop computer, but also on github CI: https://github.com/lit-robotics/libcamera-rs/pull/12

kbingham, 2023-03-08:
So https://github.com/mesonbuild/meson/issues/9721 brought this back in. The answer is simply to manually run sudo ldconfig when required as I understand it.