kbingham/libcamera #18: uvc-gadget for libcamera

kbingham, 2020-11-21:
It would be a very interesting exercise for someone to either update/port uvc-gadget to use libcamera, or to rewrite uvc-gadget from scratch making use of the c++ class library helpers for linux media available in libcamera.

groncarolo, 2022-01-27:
Hello Kieran,

A friend pointed out this PR to me :)
I might try to help.

First a dumb question: is this PR still actual?

I had a go at testing libcamera on my box and all runs OK.

I would like you to confirm me that this
git://git.ideasonboard.org/uvc-gadget.git
is the actual uvc-gadget you speak about: I ask since I found few forks not too aligned to the upstream

What would be your preference? port or jump to c++?
Given I get little to play with c++ i might prefer the latter even if it is a bit more work.
What std? 11,14 ,17?

Regards
g,

kbingham, 2022-01-27:
Yes, uvc-gadget is highly fragmented. I'd like to fix that, and make the libcamera one 'the official' one again of course ;-) I have some work that I started, but not a lot of time on it currently - certainly happy to help you work on it if you're interested.

kbingham, 2022-01-27:
We write libcamera in C++17, so I would say we would take advantage of the features there, and make as much re-use of any existing components we have already got to manage the devices.

groncarolo, 2022-01-28:
I found simple-cam, written by you, that would seem to me a very nice base from the camera related portion

I think we could hook a new std::function in in requestComplete that will forwards data to a new class
that will take care of the USB part in its own thread

Would this sound reasonable to you?

kbingham, 2022-01-28:
Can you send me an email so I can send you some references please?

groncarolo, 2022-01-28:
name . surname at gmail

kbingham, 2022-01-28:
That one will get to me, or you can use the address you'll see in my commits :-)

kbingham, 2023-02-20:
https://gitlab.freedesktop.org/camera/uvc-gadget/

closed 2023-02-20 by kbingham