kbingham/libcamera #19: bootstrap dependencies script

kbingham, 2020-11-23:
we have a growing complexity of dependencies.
It would be nice if we could wrap this up in a helper script:

> bootstrap-deps.sh [all] [cam] [qcam] [raspberrypi] [tracing] [gstreamer] [hotplug] [documentation] [tracing]

hogepodge, 2021-03-14:
As a first time user, +1 to this. The `meson build; apt install...` loop is not fun.

sfilbertf, 2023-02-23:
Hey! Is this still needed? Could I work on this as my first patch? I am interested to contribute to libcamera for GSoC 2023 :)

kbingham, 2023-02-23:
Sure, might be a helpful/useful addition - I'm not sure how we could make something generic across distributions - but it would certainly have potential to be a first contribution for sure.

sfilbertf, 2023-02-23:
Hey! I submitted my patch to the libcamera mailing list (please tell me if it did not arrive, it's my first time working with patches). I added the script to utils - it installs all required dependencies and allows installation of any optional dependencies using the syntax you wrote above.

It also has a help message to detail the arguments for the optional dependencies. For now though, it would only work for debian/ubuntu based distros (those which use apt-get as package manager). And simply doesn't do anything otherwise. Sorry for the delay, I spent a lot of time trying to find a solution to make this work generically but did not come to a compact alternative. Maybe I could take a crack at this in another PR? Thanks!

kbingham, 2023-02-23:
Yes, I think it hit the list. Thanks!

As we only specify dependencies for debian/ubuntu - I think it's reasonable that the script only handles that for now ;-) - Lets continue discussions on the list. Hopefully we can get some review comments from other community members or someone to try and test it.

kbingham, 2023-02-23:
Join the IRC/Matrix channel too if you want to get into GSoC ... it really helps to get an idea of whats going on.

sfilbertf, 2023-02-24:
I was able to find the IRC channel in webchat but wasn't able to find the matrix channel, could you tell me where I could find it? Sorry and thanks!

kbingham, 2023-02-24:
https://app.element.io/#/room/#_oftc_#libcamera:matrix.org - it's just the matrix bridge to the irc channel.