kbingham/libcamera #20: Raspberry Pi {pipeline handler/unicam} doesn't propogate stream on failures

kbingham, 2020-12-02:
With an either faulty or incorrectly connected IMX219 (RPi Camera v2) the failure case is not handled well.

When the 
In the event that the IMX219 can not be communicated with (perhaps loose cable, or faulty camera module) I experienced the following:

> pi@mercury:~/libcamera/build $ ./src/cam/cam -c1 -C10
> [0:42:32.319261681] [1863]  INFO IPAManager ipa_manager.cpp:136 libcamera is not installed. Adding '/home/pi/libcamera/build/src/ipa' to the IPA search path
> [0:42:32.358534577] [1863]  INFO Camera camera_manager.cpp:293 libcamera v0.0.11+1169-d012ba5e-dirty (2020-11-24T11:02:20+00:00)
> [0:42:32.389278693] [1864]  INFO Configuration configuration.cpp:168 Read from /usr/local/etc/libcamera/camera_sensor.json
> [0:42:32.510824486] [1864]  INFO IPAProxy ipa_proxy.cpp:122 libcamera is not installed. Loading IPA configuration from '/home/pi/libcamera/src/ipa/raspberrypi/data'
> Using camera /base/soc/i2c0mux/i2c@1/imx219@10
> [0:42:32.516979928] [1863]  INFO Camera camera.cpp:830 configuring streams: (0) 800x600-ARGB8888
> [0:42:32.517628886] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 3280x2464 fmt RG10 Score: 2592.49 (best 2592.49)
> [0:42:32.517743011] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 1920x1080 fmt RG10 Score: 2066.67 (best 2066.67)
> [0:42:32.517791825] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 1640x1232 fmt RG10 Score: 1874.49 (best 1874.49)
> [0:42:32.517838304] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 640x480 fmt RG10 Score: 2060 (best 1874.49)
> [0:42:32.517886840] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 3280x2464 fmt pRAA Score: 2092.49 (best 1874.49)
> [0:42:32.517976614] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 1920x1080 fmt pRAA Score: 1566.67 (best 1566.67)
> [0:42:32.518053685] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 1640x1232 fmt pRAA Score: 1374.49 (best 1374.49)
> [0:42:32.518123072] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 640x480 fmt pRAA Score: 1560 (best 1374.49)
> [0:42:32.518200847] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 3280x2464 fmt RGGB Score: 3592.49 (best 1374.49)
> [0:42:32.518269751] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 1920x1080 fmt RGGB Score: 3066.67 (best 1374.49)
> [0:42:32.518336805] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 1640x1232 fmt RGGB Score: 2874.49 (best 1374.49)
> [0:42:32.518402969] [1864]  INFO RPI raspberrypi.cpp:118 Mode: 640x480 fmt RGGB Score: 3060 (best 1374.49)
> [0:42:32.518500280] [1864]  INFO RPI raspberrypi.cpp:593 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 1640x1232-pRAA
> [0:42:32.581852448] [1864]  INFO RPI_S_W staggered_ctrl.cpp:36 Init ctrl 0x00980911 with delay 2
> [0:42:32.581967314] [1864]  INFO RPI_S_W staggered_ctrl.cpp:36 Init ctrl 0x009e0903 with delay 1
> [0:42:32.608911682] [1864]  INFO RPISTREAM rpi_stream.cpp:120 No buffers available for ISP Output0
> [0:42:32.609001012] [1864]  INFO RPISTREAM rpi_stream.cpp:120 No buffers available for ISP Output0
> Capture 10 frames

Which just hangs, and of course never receives frames.

dmesg shows:

> [ 2553.661661] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
> [ 2553.661675] imx219 10-0010: Failed to write reg 0x0164. error = -5
> [ 2553.661680] imx219 10-0010: imx219_start_streaming failed to set mode
> [ 2553.661722] unicam fe801000.csi: stream on failed in subdev

And that failure should have been propogated up to report the failure.

Then pressing ctrl-c on the 'hung' cam capture, adds this to the kernel logs:

> [ 2782.702834] ------------[ cut here ]------------
> [ 2782.702845] fw-clk-core already disabled
> [ 2782.702902] WARNING: CPU: 1 PID: 1864 at drivers/clk/clk.c:958 clk_core_disable+0x244/0x280
> [ 2782.702907] Modules linked in: aes_neon_blk crypto_simd cryptd rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc fuse 8021q garp stp llc imx219 joydev brcmfmac brcmutil vc4 sha256_generic libsha256 cec sg v3d drm_kms_helper bcm2835_unicam bcm2835_codec(C) cfg80211 gpu_sched v4l2_dv_timings bcm2835_v4l2(C) bcm2835_isp(C) videobuf2_vmalloc v4l2_fwnode bcm2835_mmal_vchiq(C) v4l2_mem2mem drm videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 i2c_mux_pinctrl rfkill snd_soc_core videobuf2_common drm_panel_orientation_quirks raspberrypi_ts snd_bcm2835(C) snd_compress snd_pcm_dmaengine raspberrypi_hwmon input_polldev i2c_mux dwc2 snd_pcm videodev i2c_bcm2835 mc vc_sm_cma(C) snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops rpi_backlight uio_pdrv_genirq backlight uio i2c_dev ip_tables x_tables ipv6
> [ 2782.703057] CPU: 1 PID: 1864 Comm: cam Tainted: G         C        5.4.72-v8+ #1356
> [ 2782.703063] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
> [ 2782.703071] pstate: 60000085 (nZCv daIf -PAN -UAO)
> [ 2782.703080] pc : clk_core_disable+0x244/0x280
> [ 2782.703089] lr : clk_core_disable+0x244/0x280
> [ 2782.703093] sp : ffffffc012ce39d0
> [ 2782.703099] x29: ffffffc012ce39d0 x28: 0000000040045613 
> [ 2782.703109] x27: 0000000000000000 x26: 0000000000000000 
> [ 2782.703117] x25: ffffffc012ce3c78 x24: ffffff80f1340780 
> [ 2782.703126] x23: ffffff80dd15bb80 x22: ffffff80f1340530 
> [ 2782.703134] x21: ffffff80f0bc5600 x20: ffffff80f5c36100 
> [ 2782.703141] x19: ffffff80f5c36100 x18: 0000000000000000 
> [ 2782.703149] x17: 0000000000000000 x16: 0000000000000000 
> [ 2782.703156] x15: 0000000000000000 x14: 0000000000000000 
> [ 2782.703163] x13: 0000000000000000 x12: ffffffc010fd5000 
> [ 2782.703171] x11: ffffffc010ed1000 x10: ffffffc010fd5958 
> [ 2782.703178] x9 : 0000000000000000 x8 : 0000000000000003 
> [ 2782.703186] x7 : 0000000000000173 x6 : ffffffc010fd5000 
> [ 2782.703194] x5 : 0000000000000001 x4 : 0000000000000001 
> [ 2782.703201] x3 : ffffff80f79a4150 x2 : 0000000000000007 
> [ 2782.703208] x1 : 0000000000000000 x0 : 0000000000000000 
> [ 2782.703216] Call trace:
[... 133 lines cut ...]
> [ 2782.705468] x5 : 0000000000000001 x4 : 0000000000000001 
> [ 2782.705476] x3 : ffffff80f79a4150 x2 : 0000000000000007 
> [ 2782.705483] x1 : 0000000000000000 x0 : 0000000000000000 
> [ 2782.705490] Call trace:
> [ 2782.705498]  clk_core_unprepare+0x230/0x250
> [ 2782.705506]  clk_unprepare+0x38/0x48
> [ 2782.705517]  unicam_stop_streaming+0xb0/0x158 [bcm2835_unicam]
> [ 2782.705533]  __vb2_queue_cancel+0x3c/0x280 [videobuf2_common]
> [ 2782.705548]  vb2_core_streamoff+0x34/0xc8 [videobuf2_common]
> [ 2782.705561]  vb2_streamoff+0x34/0x78 [videobuf2_v4l2]
> [ 2782.705573]  vb2_ioctl_streamoff+0x58/0x64 [videobuf2_v4l2]
> [ 2782.705612]  v4l_streamoff+0x40/0x50 [videodev]
> [ 2782.705650]  __video_do_ioctl+0x18c/0x3f0 [videodev]
> [ 2782.705688]  video_usercopy+0x214/0x748 [videodev]
> [ 2782.705726]  video_ioctl2+0x40/0x78 [videodev]
> [ 2782.705763]  v4l2_ioctl+0x68/0x90 [videodev]
> [ 2782.705782]  do_vfs_ioctl+0x6dc/0xaf8
> [ 2782.705790]  ksys_ioctl+0x88/0xb8
> [ 2782.705798]  __arm64_sys_ioctl+0x2c/0xc8
> [ 2782.705809]  el0_svc_common.constprop.0+0x7c/0x1a8
> [ 2782.705817]  el0_svc_handler+0x38/0xa8
> [ 2782.705824]  el0_svc+0x8/0x204
> [ 2782.705829] ---[ end trace ed3c627fa6775dda ]---

My initial suspicion is that the unicam driver is not propagating the stream-on failure up ?

kbingham, 2020-12-02:
It seems the Camera itself is not faulty, and when the Pi 7" touch screen is disconnected, then the camera will work again.

naushir, 2020-12-02:
See https://github.com/raspberrypi/linux/pull/3984 :)

kbingham, 2020-12-02:
And the solution to the camera and display working at the same time is to add the following to /boot/config.txt:
 "disable_touchscreen=1"

This prevents the VPU from starting the touchscreen driver.

closed 2020-12-02 by kbingham