kbingham/libcamera #21: IPU3 pipeline handler supports only 2 cameras

kbingham, 2020-12-05:
The IPU3 Pipeline handler only supports 2 cameras.

This is due to the only being 2 imgu (ISP) units, even though there are 4 CIO2 receivers.

The current implementation solves this simply by assigning a single cio2 to a single imgu and ignoring any further sensors found.

To support more than 2 cameras an implementation to share the imgu will be needed.

kbingham, 2020-12-05:
To see this limitation in the code base, see lines:
https://github.com/kbingham/libcamera/blob/master/src/libcamera/pipeline/ipu3/ipu3.cpp#L762
and
https://github.com/kbingham/libcamera/blob/master/src/libcamera/pipeline/ipu3/ipu3.cpp#L783

qzed, 2020-12-05:
I tried your quick-fix at https://github.com/linux-surface/linux-surface/issues/91#issuecomment-739335100. I can see all three cameras in the selection menu now, but I unfortunately get a segfault as soon as I select one of the cams at index 1 or 2. Index 0 works as before.

Here's the relevant section in the kernel log, if it's of any help:
```
[ 1297.397352] qcam[22209]: segfault at 58 ip 00007fb26c2e4f62 sp 00007fb26651fe00 error 4 in libcamera.so[7fb26c200000+181000]
[ 1297.397385] Code: f0 48 8b 00 48 89 c6 48 89 d7 e8 19 55 f4 ff c9 c3 90 55 48 89 e5 48 83 ec 20 48 89 7d f8 48 89 75 f0 48 89 55 e8 48 8b 45 f8 <48> 8b 48 08 48 8b 55 e8 48 8b 75 f0 48 8b 45 f8 48 89 c7 e8 56 ad
```

kbingham, 2020-12-06:
Can you run the test under gdb to get a backtrace perhaps please?

It's most likely simply that the imgu object doesn't like being shared yet.
So there will be more work to make the allocation dynamically as suggested by Laurent.

Unfortunately we won't be able to look at this anything soon at the moment. Our goals on surface are focused on getting the algorithms working to fix the colours, but we will only focus that attention on a single camera for now.

qzed, 2020-12-06:
> Can you run the test under gdb to get a backtrace perhaps please?

Sure, I'll try to find some time for that this evening.

> Unfortunately we won't be able to look at this anytime soon at the moment. Our goals on surface are focused on getting the algorithms working to fix the colours, but we will only focus that attention on a single camera for now.

No worries, I agree that getting a single camera to work has precedence.

qzed, 2020-12-06:
Alright, here's a backtrace: [gdb.log](https://github.com/kbingham/libcamera/files/5649782/gdb.log)