kbingham/libcamera #23: INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2096-50ec902a-dirty (2020-12-25T00:50:13+08:00) Camera camera not found

liuyanghan, 2020-12-29:
My camera has been installed and can shoot normally, but when I run cam, this error occurs. That's my order “cam -c camera”

kbingham, 2020-12-29:
"cam -c camera" is asking for a camera called 'camera' - and that's not an appropriate argument for the cam tool.

Please run 'cam --list', and it will show the cameras you have:

for example:

```
cam --list
[2:44:37.013148588] [27289]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2049-770df5ca
Available cameras:
1: External camera 'HP Wide Vision FHD Camera: HP W' (\_SB_.PCI0.XHC_.RHUB.HS05-5:1.0-0408:5251)
2: External camera 'HP Wide Vision FHD Camera: HP I' (\_SB_.PCI0.XHC_.RHUB.HS05-5:1.2-0408:5251)
```
Then to capture 10 frames with cam from the first camera would be:

`cam -c 1 -C 10`

But this won't show the images, so you probably want to try using qcam to see what is captured in a viewer.

liuyanghan, 2020-12-29:
Thank you very much for your prompt reply! After executing the command, no camera is available. But I can use 'raspistill -o cam.jpg'  to take photos. 

[1:16:45.845832265] [3348]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2096-50ec902a-dirty (2020-12-25T00:50:13+08:00)
Available cameras:

kbingham, 2020-12-29:
Hi, OK - so you're using a Raspberry Pi.

You need to enable the unicam peripheral to use libcamera on Raspberry Pi.
You can not use the current raspistill and libcamera at the same time.

Make sure you've followed:
https://www.raspberrypi.org/documentation/linux/software/libcamera/README.md

In particular enabling the dtoverlay for the camera

And if you have still have issues, it might be best to check the raspberry pi forums.

liuyanghan, 2020-12-29:
Thank you so much. I'll try your method now!

liuyanghan, 2020-12-30:
> Hi, OK - so you're using a Raspberry Pi.
>
> You need to enable the unicam peripheral to use libcamera on Raspberry Pi.
> You can not use the current raspistill and libcamera at the same time.
>
> Make sure [...]

Excuse me! I tried to enable the unicam as you said, but I don't know how to do it. So, I try to enabled I2C, but it still can't work. And, I had stop the raspistill . So, do you know how to enable the unicam?

kbingham, 2020-12-30:
You need to add an appropriate dtoverlay for the camera sensor you are using.

From the README.md I referenced on the raspberrypi.org site:

> Next, the /boot/config.txt file must be updated to load and use the camera driver, by adding the following lines to the bottom. Currently we also need to update the GPU's core_freq_min though this will become unnecessary in due course after further updates.
> 
> dtoverlay=imx219
> core_freq_min=250
> 
> If you are using a sensor other than the imx219 you will need to supply the alternative name here (for example, ov5647 for the V1 camera, or imx477 for the HQ Cam for which support will be available shortly).

Just to be clear, when you do this - raspistill will not work. Raspberry pi have not yet released equivalent tools to match the functionality of raspistill, but they expect to do so in the near future.

You will be able to use qcam, and cam, but they are only really test tools for experimentation with libcamera.

Do you have a particular use case in mind to use libcamera over the existing tools provided by Raspberry Pi?

liuyanghan, 2020-12-30:
I will try again.

I just want to learn ISP pipeline systematically through the libcamera open source library.

kbingham, 2020-12-30:
> I will try again.

which camera do you have?

For camera v1 you need to add:
`dtoverlay=imxov5647`

for camera v2 it's as shown in the example:
`dtoverlay=imx219`

and for the latest HD camera:
`dtoverlay=imx477`

> I just want to learn ISP pipeline systematically through the libcamera open source library.

Ok, well that's certainly a good reason to continue ;-)

Make sure you also read the documentation that Raspberry pi have produced on this topic:
https://www.raspberrypi.org/documentation/linux/software/libcamera/rpi_SOFT_libcamera_1p0.pdf

liuyanghan, 2020-12-30:
Thank you very much! I have seen this document, but I haven't read it carefully. I will take time to learn it later
My English is not very good, so in the process of installation and use, I made a lot of small mistakes.

liuyanghan, 2020-12-30:
[0:08:35.351592975] [2588]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2096-50ec902a-dirty (2020-12-30T22:41:17+08:00)
[0:08:35.430528600] [2589] ERROR V4L2 v4l2_subdevice.cpp:285 'ov5647 10-0036': Unable to get rectangle 2 on pad 0: Invalid argument
Using camera /base/soc/i2c0mux/i2c@1/ov5647@36
[0:08:35.541691170] [2588]  INFO Camera camera.cpp:890 configuring streams: (0) 800x600-ARGB8888
[0:08:35.542296315] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 2592x1944 fmt BG10 Score: 2284 (best 2284)
[0:08:35.542410342] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt BG10 Score: 2066.67 (best 2066.67)
[0:08:35.542472579] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 1296x972 fmt BG10 Score: 1717 (best 1717)
[0:08:35.542531204] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 640x480 fmt BG10 Score: 2060 (best 1717)
[0:08:35.542595719] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 640x480 fmt BA81 Score: 3060 (best 1717)
[0:08:35.542656234] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 2592x1944 fmt pBAA Score: 1784 (best 1717)
[0:08:35.542712896] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 1920x1080 fmt pBAA Score: 1566.67 (best 1566.67)
[0:08:35.542768872] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 1296x972 fmt pBAA Score: 1217 (best 1217)
[0:08:35.542824163] [2589]  INFO RPI raspberrypi.cpp:119 Mode: 640x480 fmt pBAA Score: 1560 (best 1217)
[0:08:35.542903145] [2589]  INFO RPI raspberrypi.cpp:595 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1296x972-pBAA
[0:08:35.544813119] [2589] ERROR IPARPI raspberrypi.cpp:561 Unable to find ISP control 0x009819e1
[0:08:35.544927739] [2589] ERROR IPARPI raspberrypi.cpp:302 ISP control validation failed.
[0:08:35.544973805] [2589] ERROR RPI raspberrypi.cpp:1223 IPA configuration failed!
[0:08:35.545082665] [2589] ERROR RPI raspberrypi.cpp:710 Failed to configure the IPA: -32
Failed to configure camera

This problem has been solved, thank you!
But there are other problems. I'll try and see how to solve them

By the way, There has no imx.
dtoverlay=ov5647

kbingham, 2020-12-30:
Those errors look like you might have the wrong kernel version perhaps.

It's saying that the controls it expects to find are not there.

You may need to update your kernel to the latest Raspberry Pi kernel.
I think this is done with `sudo rpi-update`.

Again, this is listed in the commands at https://www.raspberrypi.org/documentation/linux/software/libcamera/README.md

If it still doesn't work you should open a thread with the Raspberry Pi channels to support you.

Check the latest entries here too:
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=273018

But you're into areas that are supported by Raspberry Pi here.

liuyanghan, 2020-12-30:
OK! thanks a lot! I will try it.

starkfan007, 2021-03-16:
Hello kbingham,
I am using the WSL in PC to run ./cam --list, it shows:
[5:57:33.401760000] [17037]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+2407-0d0ac3cb  
Available cameras: 
So,  the camera of PC is not recognized. Could you give me some advice to solve it? Thank you very much.

kbingham, 2021-03-16:
@starkfan007 We don't support running in WSL.

Does WSL expose your camera as a UVC device?

starkfan007, 2021-03-17:
Got it. 
Thank you for your help.

starkfan007, 2021-03-17:
WSL doesn't support camera devices at this time, so I will try it on Raspberrypi 3B+.
Thank you for your instructions.