kbingham/libcamera #28: Need help with pinephone cameras bring-up (Android)

rsglobal, 2021-08-22:
I've successfully enabled all components required to run libcamera on Android (pre-12 master branch).
But during probing cameras it appears that some of mandatory controls aren't available.
Linux guys somehow uses these cameras. Are those controls really mandatory?

Please help. Here is the log: 

```
08-21 15:51:46.601   388   388 I libcamera:  INFO HAL camera3_hal.cpp:61 Initialising Android camera HAL
08-21 15:51:46.601   388   388 E libcamera: ERROR HALConfig camera_hal_config.cpp:361 Failed to open configuration file /vendor/etc/libcamera/camera_hal.yaml: No such file or directory
08-21 15:51:46.602   388   388 W libcamera:  WARN IPAManager ipa_manager.cpp:149 No IPA found in '/usr/local/lib/libcamera'
08-21 15:51:46.602   388   388 I libcamera:  INFO Camera camera_manager.cpp:294 libcamera v0.0.0
08-21 15:51:46.602   388   473 D libcamera: DEBUG Camera camera_manager.cpp:107 Starting camera manager
08-21 15:51:46.603   388   473 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "cedrus" created from /dev/media0
08-21 15:51:46.604   388   473 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: cedrus
08-21 15:51:46.604   388   473 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "sun6i-csi" created from /dev/media1
08-21 15:51:46.605   388   473 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: sun6i-csi
08-21 15:51:46.605   388   473 D libcamera: DEBUG Camera camera_manager.cpp:150 Found registered pipeline handler 'SimplePipelineHandler'
08-21 15:51:46.605   388   473 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "sun6i-csi"
08-21 15:51:46.605   388   473 D libcamera: DEBUG SimplePipeline simple.cpp:306 Found capture device sun6i-csi
08-21 15:51:47.164   388   473 D libcamera: DEBUG V4L2 v4l2_videodevice.cpp:578 /dev/video2[10:cap]: Opened device platform:csi: sun6i-video: sun6i-csi
08-21 15:51:47.164   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Exposure (0x00980911)
08-21 15:51:47.165   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Horizontal Flip (0x00980914)
08-21 15:51:47.165   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Vertical Flip (0x00980915)
08-21 15:51:47.165   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Auto Exposure (0x009a0901)
08-21 15:51:47.165   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Auto Exposure, Bias (0x009a0913)
08-21 15:51:47.165   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Analogue Gain (0x009e0903)
08-21 15:51:47.165   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Test Pattern (0x009f0903)
08-21 15:51:47.165   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'gc2145 4-003c': Control: Digital Gain (0x009f0905)
08-21 15:51:47.165   388   473 D libcamera: DEBUG CameraSensor camera_sensor.cpp:182 'gc2145 4-003c': Optional V4L2 control 0x009a0923 not supported
08-21 15:51:47.166   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:197 'gc2145 4-003c': Recommended V4L2 control 0x009a0922 not supported
08-21 15:51:47.166   388   473 E libcamera: ERROR V4L2 v4l2_subdevice.cpp:286 'gc2145 4-003c': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
08-21 15:51:47.166   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:224 'gc2145 4-003c': The PixelArraySize property has been defaulted to 1600x1200
08-21 15:51:47.166   388   473 E libcamera: ERROR V4L2 v4l2_subdevice.cpp:286 'gc2145 4-003c': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
08-21 15:51:47.166   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:235 'gc2145 4-003c': The PixelArrayActiveAreas property has been defaulted to (0x0)/1600x1200
08-21 15:51:47.166   388   473 E libcamera: ERROR V4L2 v4l2_subdevice.cpp:286 'gc2145 4-003c': Unable to get rectangle 0 on pad 0: Inappropriate ioctl for device
08-21 15:51:47.166   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:243 'gc2145 4-003c': Failed to retrieve the sensor crop rectangle
08-21 15:51:47.166   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:249 'gc2145 4-003c': The sensor kernel driver needs to be fixed
08-21 15:51:47.166   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:251 'gc2145 4-003c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
08-21 15:51:47.166   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:272 'gc2145 4-003c': Mandatory V4L2 control 0x009e0902 not available
08-21 15:51:47.166   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:272 'gc2145 4-003c': Mandatory V4L2 control 0x009f0902 not available
08-21 15:51:47.167   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:272 'gc2145 4-003c': Mandatory V4L2 control 0x009e0901 not available
08-21 15:51:47.167   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:280 'gc2145 4-003c': The sensor kernel driver needs to be fixed
08-21 15:51:47.167   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:282 'gc2145 4-003c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
08-21 15:51:47.167   388   473 E libcamera: ERROR SimplePipeline simple.cpp:1009 No valid pipeline for sensor 'gc2145 4-003c', skipping
08-21 15:51:47.167   388   473 D libcamera: DEBUG SimplePipeline simple.cpp:306 Found capture device sun6i-csi
08-21 15:51:47.167   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Contrast (0x00980901)
08-21 15:51:47.167   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Saturation (0x00980902)
08-21 15:51:47.167   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Hue (0x00980903)
08-21 15:51:47.167   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: White Balance, Automatic (0x0098090c)
08-21 15:51:47.167   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Red Balance (0x0098090e)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Blue Balance (0x0098090f)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Exposure (0x00980911)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Gain, Automatic (0x00980912)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Gain (0x00980913)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Horizontal Flip (0x00980914)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Vertical Flip (0x00980915)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Power Line Frequency (0x00980918)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Auto Exposure (0x009a0901)
08-21 15:51:47.168   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Pixel Rate (0x009f0902)
08-21 15:51:47.169   388   473 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5640 4-004c': Control: Test Pattern (0x009f0903)
08-21 15:51:47.169   388   473 D libcamera: DEBUG CameraSensor camera_sensor.cpp:182 'ov5640 4-004c': Optional V4L2 control 0x009a0923 not supported
08-21 15:51:47.169   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:197 'ov5640 4-004c': Recommended V4L2 control 0x009a0922 not supported
08-21 15:51:47.169   388   473 E libcamera: ERROR V4L2 v4l2_subdevice.cpp:286 'ov5640 4-004c': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
08-21 15:51:47.169   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:224 'ov5640 4-004c': The PixelArraySize property has been defaulted to 2592x1944
08-21 15:51:47.169   388   473 E libcamera: ERROR V4L2 v4l2_subdevice.cpp:286 'ov5640 4-004c': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
08-21 15:51:47.169   388   473 W libcamera:  WARN CameraSensor camera_sensor.cpp:235 'ov5640 4-004c': The PixelArrayActiveAr
```

rsglobal, 2021-08-22:
I found one discussion which was an attempt to solve this: https://github.com/sailfish-on-dontbeevil/droid-config-pinephone/issues/55

rsglobal, 2021-08-22:
And a set of kernel patches: https://github.com/simonschmeisser/kernel-megi/tree/5.10-libcamera

rsglobal, 2021-09-08:
```
08-21 15:51:47.166   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:272 'gc2145 4-003c': Mandatory V4L2 control 0x009e0902 not available

0x009e0902 - V4L2_CID_HBLANK

08-21 15:51:47.166   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:272 'gc2145 4-003c': Mandatory V4L2 control 0x009f0902 not available

0x009f0902 - V4L2_CID_PIXEL_RATE

08-21 15:51:47.167   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:272 'gc2145 4-003c': Mandatory V4L2 control 0x009e0901 not available

0x009e0901 - V4L2_CID_VBLANK
```

We can try to apply RPI4 experience:
https://patchwork.kernel.org/project/linux-media/cover/20201119161956.756455-1-jacopo@jmondi.org/

According to the datasheet (https://cdn.sparkfun.com/datasheets/Sensors/LightImaging/OV5640_datasheet.pdf), registers for  ov5640 and raspberry pi's ov5647 is similar (at least for hblank and vblank values)

rsglobal, 2021-09-08:
@kbingham ,

By the way, are there any plans moving libcamera to freedesktop, where other major linux graphics components are located?

rsglobal, 2021-09-08:
RPI's OV5647:
[image]

Pinephone OV5640:
[image]

kbingham, 2021-09-08:
> @kbingham ,
> 
> By the way, are there any plans moving libcamera to freedesktop, where other major linux graphics components are located?

None that I'm aware of. libcamera is a "component" of the linux-media projects https://linuxtv.org/, more closely associated with the Kernel, than the desktop.

kbingham, 2021-09-08:
> ```
> 08-21 15:51:47.166   388   473 E libcamera: ERROR CameraSensor camera_sensor.cpp:272 'gc2145 4-003c': Mandatory V4L2 control 0x009e0902 not available
>
> 0x009e0902 - V4L2_CID_HBLANK
>
> 08-21 [...]

Yes, the updates may be similar.
The three controls must be supported by the kernel driver. (And ideally upstreamed).

rsglobal, 2021-09-08:
> > @kbingham ,
> > By the way, are there any plans moving libcamera to freedesktop, where other major linux graphics components are located?
> 
> None that I'm aware of. libcamera is a "component" of the linux-media projects https://linuxtv.org/, more closely associated with the Kernel, than the desktop.

Sorry for using your personal fork for reporting project-global issues.
When I'm working via mailing lists it feels like I'm using stationary phone in the age of the smartphones. :)

kbingham, 2021-09-08:
You can also start discussions/issues on our https://bugs.libcamera.org/ bugzilla instance if you want an audience that isn't just me ;-)

rsglobal, 2021-12-16:
CC: @daaaanil81

PsychoGame, 2022-01-14:
@rsglobal,

Are you guys still busy with trying to implement the neccesary v4l2 functions into the camera drivers of the PinePhone?
Currently I'm also working on trying to get libcamera up-and-running on plain Linux, and have been in contact with developer xnux about these missing functions and if he would mentor me in trying to get these functions implemented.
He replied to me with a surprising answer that he received some camera patches lately which might implement these features.
They have been sent to me, and indeed these patches implement these functions.
Are you interested in these as well to help the GloDroid developent progress further?

Greetings Jasper

rsglobal, 2022-01-15:
Hello Jasper,

Thanks for the info. We haven't started yet, so in case patches are already available, it would be nice to start testing them.

kbingham, 2022-01-16:
Hi @PsychoGame, If there are patches to improve support for PinePhone floating around, could you consider posting them to the linux-media and libcamera mailing lists as well please to get the support reviewed for upstreaming?

kbingham, 2022-01-16:
@PsychoGame I'm not sure who xnux is, but feel free to join #libcamera on irc.oftc.net if you would like any support in getting libcamera running successfully on the pinephone too, and the support integrated upstream.

PsychoGame, 2022-01-17:
As soon as I have the chance i'll upload them.
I'm currently onboard of a shipping vessel with limited time/internet.
xnux is one of the main developers working on PinePhone kernel.
His real name is Ondrej Jirman.

> @PsychoGame I'm not sure who xnux is, but feel free to join #libcamera on
irc.oftc.net if you would like any support in getting libcamera running
successfully on the pinephone too, and the support integrated upstream.
https://ci6.googleusercontent.com/proxy/Wwslcv12-DTV6RcTOgroeO828JI8yVEqFmo6hVR4YlsiAG5xBGUJ5TTuZNhg2SEznDzo_jFHK0hUjw30IsnAUgNpgng0EFbjAwToMKMjgeMqKa7DtO93aH_-4MQqshQU2sP586LPaDaJX0dzQOzP7cJODpFROK-Tzv7g31vnb5Kq6xy3lYa5YtBFCGVr-vPcYb_Cp0rPMJkH-3xRjf7vISWnY0zsNB1SOjSotletLA=s0-d-e1-ft#https://github.com/notifications/beacon/AAZHCWD5ODSPXYN6QS4RFHDUWMDNPA5CNFSM5CTJLEOKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHRXUNSY.gif>Message
ID: ***@***.***>

rsglobal, 2022-01-18:
I also know Ondrej Jirman as @megous .  
And his main integration branch for the pinephone / PP pro is here: https://megous.com/git/linux/

PsychoGame, 2022-01-30:
@rsglobal and @kbingham,<p>

Hereby I have uploaded the promised patchsets to pastebin.<br>
These patches have not been created by me, but by a developer which goes by the name of Benjamin Schaaf.<br>
He sent them to Ondrej Jirman (Megi), which in turn has send them to me when he found out I was trying to implement the same.<br>
Eventually when Ondrej Jirman has the time to have a good look at these patchsets he'll implement them in his kernel branches as well<br>
Undermentioned are the links to the patches:<br>
- [media: ov5640: Fix mod setting semantics](https://pastebin.com/ny65VgxR)
- [media: ov5640: Add HBLANK and VBLANK controls](https://pastebin.com/GwLKLy7D)
- [media: ov5640: Improve firmware load time](https://pastebin.com/syLHNHGx)
- [media: ov5640: Fix focus commands blocking until complete](https://pastebin.com/8XJnijFR)
- [media: gc2145: Add PIXEL_RATE, HBLANK and VBLANK controls](https://pastebin.com/enC7JhwT)
- [media: sun6i-csi: Implement framesize and frameinterval enumeration](https://pastebin.com/0eqRqB4K)
- [media: sun6i-csi: Limit formats to those compatible with sub-device](https://pastebin.com/6rFDWHtf)
<p>
Hopefully you'll find them usefull in the bring-up of the PinePhone camera stack on GloDroid<br>
Greetings Jasper

rafael2k, 2022-06-22:
Can anyone confirm libcamera is working on pinephone? Not yet for me with kernel 5.15...

kbingham, 2022-06-22:
> Update: They have already been pulled in the 5.17 kernel branch

Can you update your kernel?

I know Pavel Machek has been discussing Pinephone on the libcamera mailing list in the last few days.

rafael2k, 2022-06-22:
More info.

# media-ctl -d /dev/media0 -p
Media controller API version 5.15.48

Media device information
------------------------
driver          cedrus
model           cedrus
serial          
bus info        platform:cedrus
hw revision     0x0
driver version  5.15.48

Device topology
- entity 1: cedrus-source (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video1
        pad0: Source
                -> "cedrus-proc":0 [ENABLED,IMMUTABLE]

- entity 3: cedrus-proc (2 pads, 2 links)
            type Node subtype Unknown flags 0
        pad0: Sink
                <- "cedrus-source":0 [ENABLED,IMMUTABLE]
        pad1: Source
                -> "cedrus-sink":0 [ENABLED,IMMUTABLE]

- entity 6: cedrus-sink (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video1
        pad0: Sink
                <- "cedrus-proc":1 [ENABLED,IMMUTABLE]

# media-ctl -d /dev/media1 -p
Media controller API version 5.15.48

Media device information
------------------------
driver          sun6i-csi
model           Allwinner Video Capture Device
serial          
bus info        platform:1cb0000.csi
hw revision     0x0
driver version  5.15.48

Device topology
- entity 1: sun6i-csi (1 pad, 2 links)
            type Node subtype V4L flags 0
            device node name /dev/video2
        pad0: Sink
                <- "gc2145 3-003c":0 []
                <- "ov5640 3-004c":0 [ENABLED]

- entity 5: gc2145 3-003c (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev0
        pad0: Source
                [fmt:SBGGR8_1X8/1280x960@1/30 field:none colorspace:unknown]
                -> "sun6i-csi":0 []

- entity 7: ov5640 3-004c (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev1
        pad0: Source
                [fmt:UYVY8_2X8/1280x720@1/15 colorspace:srgb xfer:srgb ycbcr:601 quantization:full-range]
                -> "sun6i-csi":0 [ENABLED]

rafael2k, 2022-06-22:
I advanced a bit here. Here I am:
./lc-compliance 
[6:03:22.948763728] [16718]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3665-bb84fc6a
[6:03:22.976465603] [16719]  WARN CameraSensor camera_sensor.cpp:212 'gc2145 3-003c': Recommended V4L2 control 0x009a0922 not supported
[6:03:22.978480561] [16719] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
[6:03:22.979363603] [16719]  WARN CameraSensor camera_sensor.cpp:239 'gc2145 3-003c': The PixelArraySize property has been defaulted to 1600x1200
[6:03:22.979433394] [16719] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
[6:03:22.979499686] [16719]  WARN CameraSensor camera_sensor.cpp:250 'gc2145 3-003c': The PixelArrayActiveAreas property has been defaulted to (0, 0)/1600x1200
[6:03:22.979599769] [16719] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 0 on pad 0: Inappropriate ioctl for device
[6:03:22.979663478] [16719]  WARN CameraSensor camera_sensor.cpp:258 'gc2145 3-003c': Failed to retrieve the sensor crop rectangle
[6:03:22.979719186] [16719]  WARN CameraSensor camera_sensor.cpp:264 'gc2145 3-003c': The sensor kernel driver needs to be fixed
[6:03:22.979773644] [16719]  WARN CameraSensor camera_sensor.cpp:266 'gc2145 3-003c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[6:03:22.979853436] [16719] ERROR CameraSensor camera_sensor.cpp:287 'gc2145 3-003c': Mandatory V4L2 control 0x009e0902 not available
[6:03:22.979915686] [16719] ERROR CameraSensor camera_sensor.cpp:287 'gc2145 3-003c': Mandatory V4L2 control 0x009f0902 not available
[6:03:22.979973686] [16719] ERROR CameraSensor camera_sensor.cpp:287 'gc2145 3-003c': Mandatory V4L2 control 0x009e0901 not available
[6:03:22.980280644] [16719] ERROR CameraSensor camera_sensor.cpp:295 'gc2145 3-003c': The sensor kernel driver needs to be fixed
[6:03:22.980343811] [16719] ERROR CameraSensor camera_sensor.cpp:297 'gc2145 3-003c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[6:03:22.981451561] [16719] ERROR SimplePipeline simple.cpp:1309 No valid pipeline for sensor 'gc2145 3-003c', skipping
[6:03:22.986176478] [16719]  WARN CameraSensor camera_sensor.cpp:212 'ov5640 3-004c': Recommended V4L2 control 0x009a0922 not supported
[6:03:22.986392019] [16719] ERROR V4L2 v4l2_subdevice.cpp:318 'ov5640 3-004c': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
[6:03:22.986487478] [16719]  WARN CameraSensor camera_sensor.cpp:239 'ov5640 3-004c': The PixelArraySize property has been defaulted to 2592x1944
[6:03:22.986555228] [16719] ERROR V4L2 v4l2_subdevice.cpp:318 'ov5640 3-004c': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
[6:03:22.986618769] [16719]  WARN CameraSensor camera_sensor.cpp:250 'ov5640 3-004c': The PixelArrayActiveAreas property has been defaulted to (0, 0)/2592x1944
[6:03:22.986690769] [16719] ERROR V4L2 v4l2_subdevice.cpp:318 'ov5640 3-004c': Unable to get rectangle 0 on pad 0: Inappropriate ioctl for device
[6:03:22.986751686] [16719]  WARN CameraSensor camera_sensor.cpp:258 'ov5640 3-004c': Failed to retrieve the sensor crop rectangle
[6:03:22.986807353] [16719]  WARN CameraSensor camera_sensor.cpp:264 'ov5640 3-004c': The sensor kernel driver needs to be fixed
[6:03:22.988926894] [16719]  WARN CameraSensor camera_sensor.cpp:266 'ov5640 3-004c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[6:03:22.990680353] [16719] ERROR CameraSensor camera_sensor.cpp:287 'ov5640 3-004c': Mandatory V4L2 control 0x009e0902 not available
[6:03:22.990761353] [16719] ERROR CameraSensor camera_sensor.cpp:287 'ov5640 3-004c': Mandatory V4L2 control 0x009e0901 not available
[6:03:22.990816561] [16719] ERROR CameraSensor camera_sensor.cpp:295 'ov5640 3-004c': The sensor kernel driver needs to be fixed
[6:03:22.990869519] [16719] ERROR CameraSensor camera_sensor.cpp:297 'ov5640 3-004c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[6:03:22.991348436] [16719] ERROR SimplePipeline simple.cpp:1309 No valid pipeline for sensor 'ov5640 3-004c', skipping

kbingham, 2022-06-22:
That looks like error messages about the kernel that are referenced as fixed by the patches listed in https://github.com/kbingham/libcamera/issues/28#issuecomment-1025098359 and - as I understand it from that post - merged in Linux 5.17.

kbingham, 2022-06-22:
@rafael2k Are you able to update the kernel? or build the latest kernel from source?

rafael2k, 2022-06-22:
I saw the patches in comment #28 and indeed tried to backport them to 5.15. No success. Did anyone mentioned success with 5.17?
I can try 5.17... but need a good source for 5.17 + pinephone patches.

kbingham, 2022-06-30:
@rafael2k can you contact Benjamin and see how he'd like to proceed, (and check if he's the author) or take over the patches yourself if he doesn't want to?

darkdragon-001, 2022-08-28:
> but need a good source for 5.17 + pinephone patches.

@rafael2k Megi (the original author of the patches) [maintains a kernel version with many patches for PinePhone](https://github.com/megous/linux) and provides binaries to download if you don't want to build from source (see [here](https://xff.cz/kernels/), `pp` stands for _PinePhone_ and `ppd` for _PinePhoneDebug_ if I remember correctly).

rafael2k, 2022-08-28:
But could you please tell me what is not working with this:
https://github.com/maemo-leste/pine64-kernel/tree/maemo/beowulf-devel
https://github.com/rafael2k/sunxi64-linux/tree/mobian-5.15
We use this kernel in Maemo and it works fine.

kbingham, 2022-08-30:
> We use this kernel in Maemo and it works fine.

"Works fine with libcamera" - or something else?

rafael2k, 2022-09-25:
Works with libcamera. But we still have a weird color conversion issue with qcam and gles driver. Without gles driver, we get:
Failed to set viewfinder format

kbingham, 2022-09-25:
Can you save a picture with qcam (with gles) and upload it here?
And capture the logs of running without gles with the LIBCAMERA_LOG_LEVELS=*:0 environment variable set please.

rafael2k, 2022-09-26:
LIBCAMERA_LOG_LEVELS=*:0 qcam -platform xcb -style=fusion                                                       
[0:30:12.356464695] [5814] DEBUG IPAModule ipa_module.cpp:320 ipa_ipu3.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_ipu3.so is not signed
[0:30:12.356812362] [5814] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_ipu3.so'
[0:30:12.357048528] [5814] DEBUG IPAModule ipa_module.cpp:320 ipa_rkisp1.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so is not signed
[0:30:12.357144278] [5814] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so'
[0:30:12.358509403] [5814] DEBUG IPAModule ipa_module.cpp:320 ipa_rpi.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so is not signed
[0:30:12.358690487] [5814] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so'
[0:30:12.358853653] [5814] DEBUG IPAModule ipa_module.cpp:320 ipa_vimc.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so is not signed
[0:30:12.358920778] [5814] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so'
[0:30:12.359063070] [5814]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:30:12.359481695] [5818] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[0:30:12.362948903] [5818] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "cedrus" created from /dev/media0
[0:30:12.363078903] [5818] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media0 due to 1 missing dependencies
[0:30:12.363554362] [5818] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media0 found
[0:30:12.363623987] [5818] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: cedrus
[0:30:12.364336487] [5818] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "sun6i-csi" created from /dev/media1
[0:30:12.364866570] [5818] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media1 due to 1 missing dependencies
[0:30:12.365288195] [5818] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media1 found
[0:30:12.365343653] [5818] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: sun6i-csi
[0:30:12.365907903] [5818] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerIPU3'
[0:30:12.366061778] [5818] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRPi'
[0:30:12.366115987] [5818] DEBUG RPI raspberrypi.cpp:1153 Unable to acquire a Unicam instance
[0:30:12.366196445] [5818] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRkISP1'
[0:30:12.366272820] [5818] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'SimplePipelineHandler'
[0:30:12.366348987] [5818] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "sun6i-csi"
[0:30:12.366580195] [5818] DEBUG SimplePipeline simple.cpp:383 Found capture device sun6i-csi
[0:30:12.366813237] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Exposure (0x00980911)
[0:30:12.366965195] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Horizontal Flip (0x00980914)
[0:30:12.367046320] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Vertical Flip (0x00980915)
[0:30:12.367189153] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Auto Exposure (0x009a0901)
[0:30:12.367267903] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Auto Exposure, Bias (0x009a0913)
[0:30:12.367448570] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Vertical Blanking (0x009e0901)
[0:30:12.367525195] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Horizontal Blanking (0x009e0902)
[0:30:12.367584737] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Analogue Gain (0x009e0903)
[0:30:12.367668570] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Pixel Rate (0x009f0902)
[0:30:12.367814278] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Test Pattern (0x009f0903)
[0:30:12.367966778] [5818] DEBUG V4L2 v4l2_device.cpp:623 'gc2145 3-003c': Control: Digital Gain (0x009f0905)
[0:30:12.372238487] [5818] DEBUG CameraSensor camera_sensor.cpp:197 'gc2145 3-003c': Optional V4L2 control 0x009a0923 not supported
[0:30:12.372360570] [5818]  WARN CameraSensor camera_sensor.cpp:212 'gc2145 3-003c': Recommended V4L2 control 0x009a0922 not supported
[0:30:12.372412987] [5818] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
[0:30:12.373244528] [5818]  WARN CameraSensor camera_sensor.cpp:239 'gc2145 3-003c': The PixelArraySize property has been defaulted to 1600x1200
[0:30:12.373321778] [5818] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
[0:30:12.373399112] [5818]  WARN CameraSensor camera_sensor.cpp:250 'gc2145 3-003c': The PixelArrayActiveAreas property has been defaulted to (0, 0)/1600x1200
[0:30:12.373488403] [5818] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 0 on pad 0: Inappropriate ioctl for device
[... cut ...]
[0:30:20.823001657] [5814] DEBUG SimplePipeline simple.cpp:808 Largest stream size is 1280x720
[0:30:20.823097407] [5814] DEBUG SimplePipeline simple.cpp:860 Picked 1280x720-SBGGR8_1X8 -> 1280x720-SBGGR8 for max stream size 1280x720
[0:30:20.823264241] [5814]  INFO Camera camera.cpp:1029 configuring streams: (0) 1280x720-SBGGR8
[0:30:20.824233282] [5818] DEBUG SimplePipeline simple.cpp:660 Link 'ov5640 3-004c':0 -> 'sun6i-csi':0 configured with format 1280x720-SBGGR8_1X8
Failed to set viewfinder format
^CExiting

kbingham, 2022-09-26:
Thanks, ok - so interesting - the simple pipeline handler is preferring RAW formats over the YUV formats (which would be much better for your usage already).

The reason it works with the GLES back end is because we have a (very basic) debayering shader to allow displaying RAW bayer formats (which is what is being configured).

Try experimenting with forcing the pixelformat:

qcam -c1 -s pixelformat=YUYV
or
qcam -c1 -s pixelformat=RGB565

kbingham, 2022-09-26:
If those formats work well, we should probably try to encourage the simple pipeline handler to have preferences on which format gets selected when there is no ISP ;-)

kbingham, 2022-09-26:
(As well as have better negotiation between cam/qcam and the default format selection)

rafael2k, 2022-09-26:
Got worse. Garbled image.
qcam -spixelformat=YUYV,width=1280,height=720
[2:52:05.862958838] [6948]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[2:52:05.869030755] [6952]  WARN CameraSensor camera_sensor.cpp:212 'gc2145 3-003c': Recommended V4L2 control 0x009a0922 not supported
[2:52:05.869151838] [6952] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
[2:52:05.869531463] [6952]  WARN CameraSensor camera_sensor.cpp:239 'gc2145 3-003c': The PixelArraySize property has been defaulted to 1600x1200
[2:52:05.869583838] [6952] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
[2:52:05.869651755] [6952]  WARN CameraSensor camera_sensor.cpp:250 'gc2145 3-003c': The PixelArrayActiveAreas property has been defaulted to (0, 0)/1600x1200
[2:52:05.869721171] [6952] ERROR V4L2 v4l2_subdevice.cpp:318 'gc2145 3-003c': Unable to get rectangle 0 on pad 0: Inappropriate ioctl for device
[2:52:05.869765463] [6952]  WARN CameraSensor camera_sensor.cpp:258 'gc2145 3-003c': Failed to retrieve the sensor crop rectangle
[2:52:05.869803630] [6952]  WARN CameraSensor camera_sensor.cpp:264 'gc2145 3-003c': The sensor kernel driver needs to be fixed
[2:52:05.869841463] [6952]  WARN CameraSensor camera_sensor.cpp:266 'gc2145 3-003c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[2:52:05.871242171] [6952]  WARN CameraSensorProperties camera_sensor_properties.cpp:174 No static properties available for 'gc2145'
[2:52:05.871306546] [6952]  WARN CameraSensorProperties camera_sensor_properties.cpp:176 Please consider updating the camera sensor properties database
[2:52:05.871349046] [6952]  WARN CameraSensor camera_sensor.cpp:411 'gc2145 3-003c': Failed to retrieve the camera location
[2:52:05.872554380] [6952]  WARN CameraSensor camera_sensor.cpp:212 'ov5640 3-004c': Recommended V4L2 control 0x009a0922 not supported
[2:52:05.872633755] [6952]  WARN CameraSensor camera_sensor.cpp:264 'ov5640 3-004c': The sensor kernel driver needs to be fixed
[2:52:05.872673421] [6952]  WARN CameraSensor camera_sensor.cpp:266 'ov5640 3-004c': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[2:52:05.873909546] [6952]  WARN CameraSensor camera_sensor.cpp:411 'ov5640 3-004c': Failed to retrieve the camera location
[2:52:16.766566635] [6948]  INFO Camera camera.cpp:1029 configuring streams: (0) 1280x720-YUYV
Using software format conversion from 1448695129
^CExiting
user@devuan-pinephone:~$ LIBCAMERA_LOG_LEVELS=*:0 qcam -spixelformat=YUYV,width=1280,height=720 -platform xcb -style=fusion
[2:52:49.759975109] [6960] DEBUG IPAModule ipa_module.cpp:320 ipa_ipu3.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_ipu3.so is not signed
[2:52:49.760378901] [6960] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_ipu3.so'
[2:52:49.760587776] [6960] DEBUG IPAModule ipa_module.cpp:320 ipa_rkisp1.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so is not signed
[2:52:49.760910067] [6960] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so'
[2:52:49.769828442] [6960] DEBUG IPAModule ipa_module.cpp:320 ipa_rpi.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so is not signed
[2:52:49.770048734] [6960] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so'
[2:52:49.770263526] [6960] DEBUG IPAModule ipa_module.cpp:320 ipa_vimc.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so is not signed
[2:52:49.770343984] [6960] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so'
[2:52:49.770494276] [6960]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[2:52:49.770897109] [6964] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[2:52:49.774448859] [6964] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "cedrus" created from /dev/media0
[2:52:49.774580776] [6964] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media0 due to 1 missing dependencies
[2:52:49.775099692] [6964] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media0 found
[2:52:49.775292817] [6964] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: cedrus
[2:52:49.775830026] [6964] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "sun6i-csi" created from /dev/media1
[2:52:49.780316651] [6964] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media1 due to 1 missing dependencies
[2:52:49.780748359] [6964] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media1 found
[2:52:49.780887567] [6964] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: sun6i-csi
[2:52:49.781388151] [6964] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerIPU3'
[2:52:49.781499567] [6964] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRPi'
[2:52:49.781637317] [6964] DEBUG RPI raspberrypi.cpp:1153 Unable to acquire a Unicam instance
[2:52:49.781707276] [6964] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRkISP1'
[... cut ...]
[2:52:56.234646487] [6964] DEBUG V4L2 v4l2_videodevice.cpp:1655 /dev/video2[14:cap]: Queueing buffer 1
[2:52:56.234766445] [6964] DEBUG V4L2 v4l2_videodevice.cpp:1655 /dev/video2[14:cap]: Queueing buffer 2
[2:52:56.309249445] [6964] DEBUG V4L2 v4l2_videodevice.cpp:1726 /dev/video2[14:cap]: Dequeuing buffer 0
[2:52:56.309592487] [6964] DEBUG Request request.cpp:122 Request(0:C:0/1:0)
[2:52:56.342735654] [6964] DEBUG V4L2 v4l2_videodevice.cpp:1726 /dev/video2[14:cap]: Dequeuing buffer 1
[2:52:56.343024320] [6964] DEBUG Request request.cpp:122 Request(1:C:0/1:0)

rafael2k, 2022-09-26:
New test files setting YUYV:
https://www.abradig.org.br/maemo-crazyness/test2.jpg
https://www.abradig.org.br/maemo-crazyness/test3.jpg

kbingham, 2022-09-26:
That looks like a stride issue. I've just realised it also says JPEG is supported.

Can you post the same with the RGB565 and  JPEG?

`qcam -c1 -s pixelformat=RGB565`

`qcam -c1 -s pixelformat=MJPEG`

I should have also already asked for the output from:

`cam -c1 -I`

kbingham, 2022-09-26:
And I see we're also dealing with two cameras on this platform. Perhaps best check both cameras too.

rafael2k, 2022-09-26:
For RGB565:
[3:04:10.649161100] [7085]  INFO Camera camera.cpp:1029 configuring streams: (0) 1280x720-RGB565
[3:04:10.650082059] [7089] ERROR SimplePipeline simple.cpp:1049 Unable to configure capture in 1280x720-RGBP (got 1280x720-RGBR)
Failed to configure camera

For MJPEG:
Same results as qcam without no argument (same as with SBGGR8 too)

cam -I for both cammeras:

Using camera /base/i2c-csi/rear-camera@4c as cam0
0: 1280x720-SBGGR8
 * Pixelformat: SBGGR8 (1280x720)-(2592x1944)/(+0,+0)
  - 1280x720
  - 1920x1080
  - 2592x1944
 * Pixelformat: RGB565 (160x120)-(2592x1944)/(+0,+0)
  - 160x120
  - 176x144
  - 320x240
  - 640x480
  - 720x480
  - 720x576
  - 1024x768
  - 1280x720
  - 1920x1080
  - 2592x1944
 * Pixelformat: SRGGB8 (1280x720)-(2592x1944)/(+0,+0)
  - 1280x720
  - 1920x1080
  - 2592x1944
 * Pixelformat: SGRBG8 (1280x720)-(2592x1944)/(+0,+0)
  - 1280x720
  - 1920x1080
  - 2592x1944
 * Pixelformat: JPEG (160x120)-(2592x1944)/(+0,+0)
  - 160x120
  - 176x144
  - 320x240
  - 640x480
  - 720x480
  - 720x576
  - 1024x768
  - 1280x720
  - 1920x1080
  - 2592x1944
 * Pixelformat: SGBRG8 (1280x720)-(2592x1944)/(+0,+0)
  - 1280x720
  - 1920x1080
  - 2592x1944
 * Pixelformat: YUYV (160x120)-(2592x1944)/(+0,+0)
  - 160x120
  - 176x144
  - 320x240
  - 640x480
  - 720x480
  - 720x576
  - 1024x768
  - 1280x720
  - 1920x1080
  - 2592x1944
 * Pixelformat: UYVY (160x120)-(2592x1944)/(+0,+0)
  - 160x120
  - 176x144
  - 320x240
  - 640x480
  - 720x480
  - 720x576
  - 1024x768
  - 1280x720
  - 1920x1080
  - 2592x1944

Using camera /base/i2c-csi/front-camera@3c as cam0
0: 1600x1200-SBGGR8
 * Pixelformat: SBGGR8 (1600x1200)-(1600x1200)/(+1,+1)
  - 1600x1200
 * Pixelformat: RGB565 (1600x1200)-(1600x1200)/(+1,+1)
  - 1600x1200
 * Pixelformat: YVYU (1600x1200)-(1600x1200)/(+1,+1)
  - 1600x1200
 * Pixelformat: YUYV (1600x1200)-(1600x1200)/(+1,+1)
  - 1600x1200
 * Pixelformat: VYUY (1600x1200)-(1600x1200)/(+1,+1)
  - 1600x1200
 * Pixelformat: UYVY (1600x1200)-(1600x1200)/(+1,+1)
  - 1600x1200

rafael2k, 2022-09-26:
Results among both cameras are similar. But lets focus on the ov5640 camera, which is rear camera, and has a much more mature driver (mainline).

kbingham, 2022-09-26:
Can you add this script to your path:
https://gist.githubusercontent.com/kbingham/34664071fcbcf237fce7f2a55f84e7a8/raw/a30873c44b5b5a6d079e28e02a88014f55538bb9/lcdebug

and then run:

`lcdebug -s qcam -c1 -s pixelformat=RGB565`

The script will just make it easier to track adding strace and log levels. It should expand to:

`LIBCAMERA_LOG_LEVELS='*:0' strace -e ioctl -f -- qcam -c1 -s pixelformat=RGB565`

But perhaps given the iterations here already - would it be possible to set up some live / remote debugging? Could we join a video call to go through this in realtime perhaps?

If you can join the IRC/Matrix channel (oftc.net #libcamera, https://matrix.to/#/#_oftc_#libcamera:matrix.org) we can organise getting on a call to figure this out.

rafael2k, 2022-09-27:
I'll show up in irc.  I'll run the tests you suggest also, later in the day I publish the results.

rafael2k, 2022-09-27:
Here the output of
lcdebug -s qcam -c "/base/i2c-csi/rear-camera@4c" -s pixelformat=RGB5652
https://www.abradig.org.br/maemo-crazyness/debug.txt

rafael2k, 2022-09-27:
Latest tests with YUYV in qcam:
https://www.abradig.org.br/maemo-crazyness/test4.jpg
Using just cam and -spixelformat=YUYV,width=1280,height=720 (and then converting with ffmpeg):
https://www.abradig.org.br/maemo-crazyness/test5.jpg

closed 2022-10-17 by rsglobal

kbingham, 2022-10-17:
@rafael2k as rsglobal has closed this issue, please either raise any remaining issues on the libcamera development mailing list, or bugs.libcamera.org. (You can open another issue in this repository, but please remember that only I follow this repo. It's my personal fork - it's not an official libcamera repository).

rafael2k, 2022-10-17:
a last comment: the camera patches listed by @PsychoGame I uploaded here:
https://www.abradig.org.br/maemo-crazyness/camera_patches/

kbingham, 2022-10-17:
Ok - those patches need to get posted to the linux kernel though. Not libcamera. (Or check what the state is compared to an upstream linux kernel). Thanks for tracking them.