kbingham/libcamera #31: Android: RPI4: Need help with log analysis

rsglobal, 2021-11-08:
Currently I have no camera picture and this:
```
10-07 14:42:18.486   413  1588 E Camera3-Stream: getBuffer: wait for output buffer return timed out after 8000ms (max_buffers 3)
```
message from the framework.

I'm on the latest libcamera & rpi-kernel v5.10.y

I will appreciate any ideas and suggestions where to look further.

```
adb logcat | grep libcam
01-01 00:00:23.386   360   360 I libcamera:  INFO HAL camera3_hal.cpp:61 Initialising Android camera HAL
01-01 00:00:23.416   360   360 D libcamera: DEBUG HALConfig camera_hal_config.cpp:378 '/base/soc/i2c0mux/i2c@1/ov5647@36' (0)[0]
01-01 00:00:23.544   360   360 D libcamera: DEBUG IPAModule ipa_module.cpp:320 ipa_rpi.so: IPA module /vendor/lib64/libcamera//ipa_rpi.so is not signed
01-01 00:00:23.544   360   360 D libcamera: DEBUG IPAManager ipa_manager.cpp:241 Loaded IPA module '/vendor/lib64/libcamera//ipa_rpi.so'
01-01 00:00:23.549   360   360 I libcamera:  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
01-01 00:00:23.550   360   442 D libcamera: DEBUG Camera camera_manager.cpp:106 Starting camera manager
01-01 00:00:23.551   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media2
01-01 00:00:23.554   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: bcm2835-codec
01-01 00:00:23.554   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "unicam" created from /dev/media0
01-01 00:00:23.555   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: unicam
01-01 00:00:23.556   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media1
01-01 00:00:23.578   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: bcm2835-isp
01-01 00:00:23.578   360   442 D libcamera: DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'SimplePipelineHandler'
01-01 00:00:23.578   360   442 D libcamera: DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRPi'
01-01 00:00:23.579   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "unicam"
01-01 00:00:23.579   360   442 D libcamera: DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "bcm2835-isp"
01-01 00:00:23.581   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: White Balance, Automatic (0x0098090c)
01-01 00:00:23.581   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Exposure (0x00980911)
01-01 00:00:23.581   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Gain, Automatic (0x00980912)
01-01 00:00:23.582   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Auto Exposure (0x009a0901)
01-01 00:00:23.582   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Camera Orientation (0x009a0922)
01-01 00:00:23.585   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Camera Sensor Rotation (0x009a0923)
01-01 00:00:23.586   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Vertical Blanking (0x009e0901)
01-01 00:00:23.586   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Horizontal Blanking (0x009e0902)
01-01 00:00:23.586   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Analogue Gain (0x009e0903)
01-01 00:00:23.586   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 'ov5647 10-0036': Control: Pixel Rate (0x009f0902)
01-01 00:00:23.591   360   442 D libcamera: DEBUG CameraSensor camera_sensor.cpp:318 'ov5647 10-0036': No static test pattern map for 'ov5647'
01-01 00:00:23.592   360   442 D libcamera: DEBUG IPAManager ipa_manager.cpp:299 IPA module /vendor/lib64/libcamera//ipa_rpi.so signature is not valid
01-01 00:00:23.601   360   442 D libcamera: DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /vendor/lib64/libcamera//ipa_rpi.so
01-01 00:00:24.671   360   442 D libcamera: DEBUG RPiBlackLevel black_level.cpp:41  Read black levels red 1024 green 1024 blue 1024
01-01 00:00:24.672   360   442 D libcamera: DEBUG RPiAgc agc.cpp:191 Agc
01-01 00:00:24.672   360   442 D libcamera: DEBUG RPiAgc agc.cpp:144 AgcConfig
01-01 00:00:24.675   360   442 D libcamera: DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cr calibration for ct 3000
01-01 00:00:24.676   360   442 D libcamera: DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cr calibration for ct 5000
01-01 00:00:24.678   360   442 D libcamera: DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cr calibration for ct 6500
01-01 00:00:24.678   360   442 D libcamera: DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cb calibration for ct 3000
01-01 00:00:24.679   360   442 D libcamera: DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cb calibration for ct 5000
01-01 00:00:24.680   360   442 D libcamera: DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cb calibration for ct 6500
01-01 00:00:24.682   360   442 D libcamera: DEBUG RPiSharpen sharpen.cpp:45 Read threshold 1 strength 1 limit 1
01-01 00:00:24.722   360   442 D libcamera: DEBUG V4L2 v4l2_videodevice.cpp:588 /dev/video0[15:cap]: Opened device platform:fe801000.csi: unicam: unicam
01-01 00:00:24.723   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Red Balance (0x0098090e)
01-01 00:00:24.723   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Blue Balance (0x0098090f)
01-01 00:00:24.723   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Colour Correction Matrix (0x009819e1)
01-01 00:00:24.724   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Lens Shading (0x009819e2)
01-01 00:00:24.724   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Black Level (0x009819e3)
01-01 00:00:24.724   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Green Equalisation (0x009819e4)
01-01 00:00:24.724   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Gamma (0x009819e5)
01-01 00:00:24.724   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Denoise (0x009819e6)
01-01 00:00:24.725   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Sharpen (0x009819e7)
01-01 00:00:24.725   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Defective Pixel Correction (0x009819e8)
01-01 00:00:24.725   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Colour Denoise (0x009819e9)
01-01 00:00:24.726   360   442 D libcamera: DEBUG V4L2 v4l2_device.cpp:610 /dev/video13[16:cap]: Control: Digital Gain (0x009f0905)
01-01 00:00:24.726   360   442 D libcamera: DEBUG V4L2 v4l2_videodevice.cpp:588 /dev/video13[16:out]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
01-01 00:00:24.726   360   442 D libcamera: DEBUG V4L2 v4l2_videodevice.cpp:588 /dev/video14[17:cap]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
01-01 00:00:24.727   360   442 D libcamera: DEBUG V4L2 v4l2_videodevice.cpp:588 /dev/video15[18:cap]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
01-01 00:00:24.727   360   442 D libcamera: DEBUG V4L2 v4l2_videodevice.cpp:588 /dev/video16[19:cap]: Opened device platform:bcm2835-isp: bcm2835-isp: bcm2835-isp
01-01 00:00:24.727   360   442 D libcamera: DEBUG DelayedControls delayed_controls.cpp:99 Set a delay of 2 and priority write flag 1 for Vertical Blanking
01-01 00:00:24.727   360   442 D libcamera: DEBUG DelayedControls delayed_controls.cpp:99 Set a delay of 2 and priority write flag 0 for Exposure
[... 1565 lines cut ...]
10-07 14:37:49.500   360   442 D libcamera: DEBUG DelayedControls delayed_controls.cpp:237 frame 49 started
10-07 14:37:49.501   360   442 D libcamera: DEBUG DelayedControls delayed_controls.cpp:285 Queue is empty, auto queue no-op.
10-07 14:37:49.564   360   442 D libcamera: DEBUG RPI raspberrypi.cpp:1281 frame start 50
[... cut ...]
10-07 14:37:49.884   360   442 D libcamera: DEBUG DelayedControls delayed_controls.cpp:285 Queue is empty, auto queue no-op.
10-07 14:37:49.948   360   442 D libcamera: DEBUG RPI raspberrypi.cpp:1281 frame start 56
10-07 14:37:49.948   360   442 D libcamera: DEBUG DelayedControls delayed_controls.cpp:237 frame 56 started
10-07 14:37:49.948   360   442 D libcamera: DEBUG DelayedControls delayed_controls.cpp:285 Queue is empty, auto queue no-op.
10-07 14:37:50.012   360   442 D libcamera: DEBUG RPI raspberrypi.cpp:1281 frame start 57
10-07 14:37:50.012   360   442 D libcamera: D

kbingham, 2021-11-08:
Could you post this question to the libcamera mailling list please?

On this repository you get an audience of just *me*.
On the libcamera mailing list you get the other libcamera developers, as well as the developers who maintain the raspberry pi pipeline handler and IPA at Raspberry Pi.

rsglobal, 2021-11-09:
While debugging, got a state in which stream started for some reason. So I'm really close to get libcamera+android+rpi fully functional.

[image]

kbingham, 2021-11-09:
Great! Did you identify what the change was that enabled the streaming?

rsglobal, 2021-11-09:
No I haven't fixed it yet. It just looks like restarting camera app did the thing.

uajain, 2021-11-09:
Hi @rsglobal 

Can you briefly explain your OS setup with AOSP/android+Rpi ? I am interested.

rsglobal, 2021-11-09:
Hi @uajain ,

> Can you briefly explain your OS setup with AOSP/android+Rpi ? I am interested.

Not sure I can explain all setup briefly.

Here is our project: https://github.com/GloDroid/glodroid_manifest

You can look into glodroid.xml file to see the difference from AOSP.

rsglobal, 2021-11-09:
Green row fixed by removing buffer height alignment in the gralloc.
[image]

rsglobal, 2021-11-09:
Color correction works pretty well.

rsglobal, 2021-11-12:
I've applied [this](https://patchwork.libcamera.org/cover/14586/) series and picture freezes again.
I'm attaching the logs:
[libcamera_v2.log](https://github.com/kbingham/libcamera/files/7529005/libcamera_v2.log)

kbingham, 2021-11-12:
@naushir It seems to have crashed in `RPiCameraData::ispOutputDequeue`
```
10-07 14:39:27.028  2242  2242 F DEBUG   : backtrace:
10-07 14:39:27.028  2242  2242 F DEBUG   :       #00 pc 0000000000097c04  /apex/com.android.vndk.v31/lib64/libc++.so (std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)+28) (BuildId: 0db72613bf270f11efab0a156c4fe541)
10-07 14:39:27.029  2242  2242 F DEBUG   :       #01 pc 00000000000bf378  /vendor/lib64/libcamera.so (libcamera::RPiCameraData::ispOutputDequeue(libcamera::FrameBuffer*)+192) (BuildId: 4282ee1d889803300eeb9931b3b3b9e6)
10-07 14:39:27.029  2242  2242 F DEBUG   :       #02 pc 00000000000c3680  /vendor/lib64/libcamera.so (BuildId: 4282ee1d889803300eeb9931b3b3b9e6)
10-07 14:39:27.029  2242  2242 F DEBUG   :       #03 pc 00000000000ae300  /vendor/lib64/libcamera.so (BuildId: 4282ee1d889803300eeb9931b3b3b9e6)
10-07 14:39:27.029  2242  2242 F DEBUG   :       #04 pc 00000000000af37c  /vendor/lib64/libcamera.so (BuildId: 4282ee1d889803300eeb9931b3b3b9e6)
10-07 14:39:27.029  2242  2242 F DEBUG   :       #05 pc 00000000000123c8  /vendor/lib64/libcamera-base.so (BuildId: 5609a3280ab3276ccad765984e985601)
```

Is there an obvious path to there which may have failed to dequeue a buffer perhaps? Or something happening due to the v4l2 buffer cache issue maybe?

rsglobal, 2021-11-12:
@kbingham ,

I've appended full crash and addr2line in the end of log file.

naushir, 2021-11-15:
> Is there an obvious path to there which may have failed to dequeue a buffer perhaps? Or something happening due to the v4l2 buffer cache issue maybe?

The only think I can see is if ``stream`` is null, there is an assert that checks for this.  This can only happen if a buffer is dequeued by the driver that pipeline handler cannot recognise.  Again, this may be something to do with the v4l2 cache being under allocated, not too sure.

From the logs, this does not seem to crash immediately, but after a few restarts, is that correct?

naushir, 2021-11-15:
It would be worth running with this snippet and see if you hit the warning message:

```
diff --git a/src/libcamera/pipeline/raspberrypi/raspberrypi.cpp b/src/libcamera/pipeline/raspberrypi/raspberrypi.cpp
index 4f6c699a4379..815ba860b126 100644
--- a/src/libcamera/pipeline/raspberrypi/raspberrypi.cpp
+++ b/src/libcamera/pipeline/raspberrypi/raspberrypi.cpp
@@ -1551,8 +1551,10 @@ void RPiCameraData::ispOutputDequeue(FrameBuffer *buffer)
                }
        }
 
-       /* The buffer must belong to one of our ISP output streams. */
-       ASSERT(stream);
+       if (!stream) {
+               LOG(RPI, Error) << "Unrecongised stream!!");
+               return;
+       }
 
        LOG(RPI, Debug) << "Stream " << stream->name() << " buffer complete"
                        << ", buffer id " << index
```

rsglobal, 2021-11-16:
Attaching new logs: It looks like this time it didn't crash for some reason. But I can't see added fatal anywhere.

[libcam_3.log](https://github.com/kbingham/libcamera/files/7548358/libcam_3.log)

naushir, 2021-11-17:
```
10-07 14:37:46.786   381   441 E libcamera: ERROR RPISTREAM rpi_stream.cpp:239 Failed to queue buffer for ISP Output1
10-07 14:37:46.786   381   443 D libcamera: DEBUG RPiAwb awb.cpp:522 At t 6438.89 r 0.650454 b 0.895426: 0.696849
10-07 14:37:46.786   381   441 D libcamera: DEBUG RPISTREAM rpi_stream.cpp:234 Queuing buffer 1048579 for ISP Output1
10-07 14:37:46.786   381   441 E libcamera: ERROR RPISTREAM rpi_stream.cpp:239 Failed to queue buffer for ISP Output1
```

This error shows the same buffer cache issue again,  where seem to be running out of buffer slots.  There seems to be quite a bit of variability in the failure cases from what I can tell, and without knowing anything about what the Android wrapper is doing, I can't really suggest much more to try.

When I am able to, perhaps this something I can recreate on my end to debug.  Do you have a set of instructions for setting your environment up so I can reproduce this?

rsglobal, 2021-11-17:
> When I am able to, perhaps this something I can recreate on my end to debug. Do you have a set of instructions for setting your environment up so I can reproduce this?

Building AOSP has advanced hardware requirements:
Minimum: 16GB of RAM (in this case all other apps must be closed) or 32gb of RAM and 300-400GB free disk space.

In case you have all this I can prepare release/instructions for you.

naushir, 2021-11-17:
Even better, would you be able to create a disk image so I can flash it directly?

With all the other things going on, I am not sure when I will be able to get to this though.

rsglobal, 2021-11-17:
> Even better, would you be able to create a disk image so I can flash it directly?

Yes, but you won't be able to rebuild libcamera.

kbingham, 2021-11-17:
At the least, I have the build capabilities for this, so instructions are probably very useful to publish somewhere if that's possible.

But I wonder if we can try to isolate the issue and replicate it in a simpler case somehow.

@rsglobal Do I recall seeing that you said the buffers /always/ use a different FD, and so are blowing through the cache because that's how we match, and that we need to check the inodes instead?

Can we verify this by printing the fd and inodes to see them get used? If the v4l2buffercache is at fault here, is that all we need to fix?

rsglobal, 2021-11-17:
> At the least, I have the build capabilities for this, so instructions are probably very useful to publish somewhere if that's possible.

Sure, let me just finalize v0.7.0 GloDroid release. All instructions are already [HERE](https://github.com/GloDroid/glodroid_manifest#fetching-android-sources)

> @rsglobal Do I recall seeing that you said the buffers /always/ use a different FD, and so are blowing through the cache because that's how we match, and that we need to check the inodes instead?

I just assumed that buffers may use different fd if they are received from CLIENT. But honestly I do not understand what that cache are doing, and what is the info stored there and for what purpose. 

> Can we verify this by printing the fd and inodes to see them get used? If the v4l2buffercache is at fault here, is that all we need to fix?

Yes, ping me later please for that. (in a few weals).

kbingham, 2021-11-17:
V4L2 uses separate structures to represent the internal buffers, and we give it a dma-buf handle ( the fd ). 
If those are not kept associated, so the same v4l2 buffer is not used for the same underlying buffer, there is a performance penalty.

If there really is a distinct different buffer each time, we may blow through the cache, but we should instead take the performance penalty and re-use the oldest v4l2 structure (which it doesn't do now).

However, if the fd changes, but maps to the same underlying inode (same actual dma-buf) then we should use that to map the cache entry.

To know what's happening in this use case, printing all that information before it stalls would really help.
I can't replicate the issue, as both locally and in chromeos where we use the android layer, the fd's are re-used so we don't exhaust the cache. This is of course 'assuming' that the underlying issue is related to that, so again - printing out the fd's to see them would really help confirm/deny the theory.

kbingham, 2021-11-17:
Also, I believe we map on the FD rather than the inode currently, as otherwise - mapping on the inode would otherwise incur an extra system call to get the inode.... but perhaps it becomes unavoidable ?

rsglobal, 2021-11-24:
Hi, now you can try Android-12 on RPI4 with the libcamera.
Images are available under assets list:
https://github.com/GloDroid/glodroid_manifest/releases/tag/v0.7.0

To build from sources please follow this manual: https://github.com/GloDroid/glodroid_manifest#fetching-android-sources

kbingham, 2022-10-17:
Is this still an issue ?

closed 2022-10-17 by rsglobal