kbingham/libcamera #34: ERROR: *** no cameras available *** - OV9281

AndreGuimRua, 2021-11-23:
Hello,

I'm trying to get libcamera work on a Raspberry 4 with an innomaker OV9281 camera
OS
`Raspberry Bullseye 5.10.63-v7l+`

vcgencmd version
`
Oct 29 2021 10:47:33
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
`
dmesg | grep -i OV9281 returns nothing
I've also checked with another Pi 4 and flex cable with the same result, but I'm able to open the camera with VLC and get the video output

It works fine with an IMX477 camera

Best regards

kbingham, 2021-11-23:
Hi,

Have you constructed and enabled an appropriate overlay for the OV9281 in your ? Or is it supported by Raspberry Pi?

Have you updated your /boot/config.txt to specify the correct (and available?) dto file for your camera?

kbingham, 2021-11-23:
You said you are able to open the camera with VLC and get video output ... what is that from? The OV9281? If so - it sounds like you are mixing and matching the 'old' camera stack and the 'new'. They are not both compatible.

AndreGuimRua, 2021-11-23:
Yeah, the OV9281 feed on VLC was with a previous stack, it was just to check that the camera was OK

I thought that with the new feature of camera_auto_detect on Bullseye specifying the overlay wasn't required anymore
Added `dtoverlay=OV9281` at the bottom of config.txt and it seems that's working now

Thanks @kbingham :)

closed 2021-11-23 by AndreGuimRua

rsglobal, 2021-11-23:
Hi,
You can try `camera_auto_detect=1` option instead of `dtoverlay=`:

AndreGuimRua, 2021-11-23:
> 
´camera_auto_detect=1´ already was like that and didn't work, specifying the overlay did

rsglobal, 2021-11-23:
`camera_auto_detect=1` requires recently released firmware as far as I know.

kbingham, 2021-11-23:
Oh - nice, I didn't know about camera_auto_detect=1. I'll have to ask the RPi guys how that gets handled.

rsglobal, 2021-11-23:
https://github.com/RPi-Distro/pi-gen/blob/master/stage1/00-boot-files/files/config.txt#L55

AndreGuimRua, 2021-11-23:
> `camera_auto_detect=1` requires recently released firmware as far as I know.

Yeah, I'm using the latest verison of Raspberry OS (October 30th 2021) 2021-05-07-raspios-buster-armhf-full
[https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-11-08/2021-10-30-raspios-bullseye-armhf-full.zip](url)

So it's the latest atm : )