kbingham/libcamera #35: Use libcamera in docker container

ambrosekwok, 2021-12-18:
Hello,

I try to use libcamera in docker container, but I got below error messages

```
root@29f288ba1870:/# gst-device-monitor-1.0
Probing devices...

(gst-device-monitor-1.0:9): GStreamer-CRITICAL **: 07:38:52.786: Trying to stop a GstDeviceProvider libcameraprovider0 which is already stopped
Failed to start device monitor!

root@29f288ba1870:/# gst-device-monitor-1.0 Video
Probing devices...

[1:29:45.684751100] [13]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3369-8ff5a8d5
No devices found!
```

Here is dockerfile used:
```
FROM ubuntu:20.04

ENV DEBIAN_FRONTEND noninteractive

RUN apt-get update && apt-get --no-install-recommends install -y python3-pip git
RUN apt-get install -y cmake libgnutls28-dev openssl libboost-dev
RUN pip3 install pyyaml ninja meson jinja2 ply
RUN apt-get install -y libgstreamer1.0-0 libgstreamer1.0-dev gstreamer1.0-dev gstreamer1.0-libav gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-plugins-base-apps libgstreamer-plugins-base1.0-dev
    
RUN apt-get install -y libgnutls28-dev openssl libudev-dev libboost-dev qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5 \
    && pip3 install pyyaml ply jinja2
RUN git clone https://git.libcamera.org/libcamera/libcamera.git \
    && cd libcamera \
    && meson build --buildtype=release -Dpipelines=raspberrypi -Dipas=raspberrypi -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled \
    && ninja -C build install
RUN cp /usr/local/lib/aarch64-linux-gnu/gstreamer-1.0/* /usr/lib/aarch64-linux-gnu/gstreamer-1.0/

CMD [ "gst-device-monitor-1.0", "Video"]
```

Command used to run the container:
`docker run -it --rm --privileged --device /dev/vchiq:/dev/vchiq --device /dev/dri:/dev/dri -v /dev/bus/usb:/dev/bus/usb -v /dev/video0:/dev/video0 -v /dev/video1:/dev/video1 libcamera bash`

The libcamera work fine on the host with Raspberry Pi OS, but failed inside container.
Any advise to make libcamera work in docker container? Thank you.

Best regards

kbingham, 2021-12-19:
You'll need to pass in at least the /dev/media* devices and /dev/v4l-subdev* devices into the docker container too

ambrosekwok, 2021-12-21:
Modified the docker run command as below, but still no working.
`docker run -it --rm --privileged --device /dev/vchiq:/dev/vchiq --device /dev/dri:/dev/dri -v /dev/bus/usb:/dev/bus/usb -v /dev/video0:/dev/video0 -v /dev/video1:/dev/video1 -v /dev/v4l-subdev0:/dev/v4l-subdev0 -v /dev/media0:/dev/media0  -v /dev/media1:/dev/media1  -v /dev/media2:/dev/media2 libcamera bash`

kbingham, 2021-12-21:
Can you post a full log of running cam --list in the container?

LIBCAMERA_LOG_LEVELS=*:0 cam --list

(Or the gst-device-monitor-1.0 Video with full logging too)

ambrosekwok, 2021-12-21:
Please find the log as below, thank you.
```
root@5d512b2b9f00:/# LIBCAMERA_LOG_LEVELS=*:0 cam --list
[0:28:24.640824547] [82] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so is signed
[0:28:24.641193377] [82] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so'
[0:28:24.641402893] [82]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3369-8ff5a8d5
[0:28:24.641836371] [83] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[0:28:24.655992630] [83] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerUVC'
[0:28:24.656230961] [83] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRPi'
[0:28:24.656528866] [83] DEBUG RPI raspberrypi.cpp:1060 Unable to acquire a Unicam instance
Available cameras:
root@5d512b2b9f00:/#
```

gst-device-monitor-1.0 Video --gst-debug-level=9
[gst-device-monitor-1.0.log](https://github.com/kbingham/libcamera/files/7755432/gst-device-monitor-1.0.log)

kbingham, 2021-12-21:
I believe your issue is to do with udev not functioning as we expect inside the container.

We already have a means to fall back to using sysfs to enumerate devices, so if you remove libudev-dev from your dependencies it should use that.

I've tried it locally on an X86 and managed to get devices listing there, so it's worth a go, and might be enough of a workaround.

For a real solution you might want to try investigating how or why udev isn't doing what is expected inside the container, but maybe not using it is enough.

kbingham, 2021-12-23:
Any luck here? Did this solve your issues? Or did you find a way to get udev running in the container?

ambrosekwok, 2021-12-23:
Thank you, but I still cannot let it work, after removed libudev-dev from container. I got below messages:
```
root@f36e68eaec44:/# LIBCAMERA_LOG_LEVELS=*:0 cam --list
[0:22:23.454414826] [11] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so is signed
[0:22:23.455219997] [11] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so'
[0:22:23.455772796] [11]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3369-8ff5a8d5
[0:22:23.456940246] [12] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[0:22:23.458117622] [12] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "unicam" created from /dev/media2
[0:22:23.460404115] [12] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: unicam
[0:22:23.461234989] [12] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media0
[0:22:23.463549815] [12] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: bcm2835-isp
[0:22:23.464133836] [12] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media1
[0:22:23.465782363] [12] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: bcm2835-codec
[0:22:23.466240106] [12] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRPi'
[0:22:23.466566922] [12] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "unicam"
[0:22:23.466830313] [12] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "bcm2835-isp"
[0:22:23.467480742] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: White Balance, Automatic (0x0098090c)
[0:22:23.467677150] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Exposure (0x00980911)
[0:22:23.467768150] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Gain, Automatic (0x00980912)
[0:22:23.467857225] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Auto Exposure (0x009a0901)
[0:22:23.467976800] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Camera Orientation (0x009a0922)
[0:22:23.469368584] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Camera Sensor Rotation (0x009a0923)
[0:22:23.469803272] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Vertical Blanking (0x009e0901)
[0:22:23.470098569] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Horizontal Blanking (0x009e0902)
[0:22:23.470264200] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Analogue Gain (0x009e0903)
[0:22:23.470502923] [12] DEBUG V4L2 v4l2_device.cpp:611 'ov5647 10-0036': Control: Pixel Rate (0x009f0902)
[0:22:23.473260677] [12] DEBUG CameraSensor camera_sensor.cpp:317 'ov5647 10-0036': V4L2_CID_TEST_PATTERN is not supported
[0:22:23.473617401] [12] ERROR CameraSensor camera_sensor.cpp:535 'ov5647 10-0036': Camera sensor does not support test pattern modes.
[0:22:23.488476177] [12] DEBUG IPAManager ipa_manager.cpp:298 IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so signature is valid
[0:22:23.489104884] [12] DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so
[0:22:23.498922710] [12] DEBUG RPiBlackLevel black_level.cpp:41  Read black levels red 1024 green 1024 blue 1024
[0:22:23.499874992] [12] DEBUG RPiAgc agc.cpp:191 Agc
[0:22:23.499968919] [12] DEBUG RPiAgc agc.cpp:144 AgcConfig
[0:22:23.501971262] [12] DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cr calibration for ct 3000
[0:22:23.502754506] [12] DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cr calibration for ct 5000
[0:22:23.503491936] [12] DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cr calibration for ct 6500
[0:22:23.504382496] [12] DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cb calibration for ct 3000
[0:22:23.505304130] [12] DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cb calibration for ct 5000
[0:22:23.506261172] [12] DEBUG RPiAlsc alsc.cpp:118 Read calibrations_Cb calibration for ct 6500
[0:22:23.506927767] [12] DEBUG RPiSharpen sharpen.cpp:45 Read threshold 1 strength 1 limit 1
[0:22:23.507576641] [12]  WARN RPI raspberrypi.cpp:1145 Mismatch between Unicam and CamHelper for embedded data usage!
[0:22:23.538610107] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: White Balance, Automatic (0x0098090c)
[0:22:23.538849145] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Exposure (0x00980911)
[0:22:23.538983128] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Gain, Automatic (0x00980912)
[0:22:23.539684149] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Auto Exposure (0x009a0901)
[0:22:23.539754724] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Camera Orientation (0x009a0922)
[0:22:23.539801780] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Camera Sensor Rotation (0x009a0923)
[0:22:23.539841854] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Vertical Blanking (0x009e0901)
[0:22:23.539882262] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Horizontal Blanking (0x009e0902)
[0:22:23.539918151] [12] DEBUG V4L2 v4l2_device.cpp:611 /dev/video0[15:cap]: Control: Analogue Gain (0x009e0903)
[... cut ...]
[0:22:23.543008758] [12] DEBUG V4L2 v4l2_videodevice.cpp:1192 /dev/video0[15:cap]: 0 buffers requested.
[0:22:23.546712758] [12] ERROR RPI raspberrypi.cpp:1074 Failed to register camera: -22
Available cameras:
root@f36e68eaec44:/#
```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

root@30eb5d622e27:/#  uname -a
Linux 30eb5d622e27 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 aarch64 aarch64 GNU/Linux
```

kbingham, 2021-12-23:
That's probably something to report/take up with the RPi developers perhaps at that point. You could post that on bugs.libcamera.org or post it to the libcamera-devel mailinglist, or contact RPi through their forums.

bugsyb, 2022-03-09:
> Thank you, but I still cannot let it work, after removed libudev-dev from container. I got below messages:
> 
@ambrosekwok  - did you manage to get it resolved and got the libcamera working in container? Hope is that by now you've a solution.

I'm heading same way, hence asking.

Thanks!

kbingham, 2022-03-10:
See : https://github.com/raspberrypi/libcamera-apps/issues/270#issuecomment-1059372626
> i ran udevadm control --reload in the container

That may help

kbingham, 2022-03-10:
https://github.com/raspberrypi/libcamera-apps/issues/270#issuecomment-1059372626 confirms libcamera can run in docker environments, so I'll close this now.

closed 2022-03-10 by kbingham