kbingham/libcamera #36: RFU: Check for pixelformat availability before accessing to avoid crash

rsglobal, 2021-12-24:
RFU - Request for upstream :)

@kbingham ,

Hi Kieran, 

Could you please send this patch to ML if relevant for other.

```diff
--- a/src/libcamera/pipeline/uvcvideo/uvcvideo.cpp
+++ b/src/libcamera/pipeline/uvcvideo/uvcvideo.cpp
@@ -122,6 +122,9 @@ CameraConfiguration::Status UVCCameraConfiguration::validate()
        const Size size = cfg.size;
 
        const std::vector<PixelFormat> pixelFormats = formats.pixelformats();
+       if (pixelFormats.empty())
+               return Invalid;
+
        auto iter = std::find(pixelFormats.begin(), pixelFormats.end(), pixelFormat);
        if (iter == pixelFormats.end()) {
                cfg.pixelFormat = pixelFormats.front();
```

kbingham, 2021-12-24:
Looks same initially, can you provide since more information regarding the crash that occurs and what you tried when it happened please?

rsglobal, 2021-12-24:
I decided to use libcamera for external cameras support for Android instead of external camera HAL built-in into android.

Built-in HAL doesn't support a set of cheap cameras, which was my motivation to use libcamera.

But it turned-out that enabling of uvcpipeline isn't enough and there are a lot of work need to be done to accomplish my goal.

I started to put stubs in a places which caused pipeline initialization to fail expecting to get at least any results and at some point faced this crash. With this patch I can see `Camera configuration invalid` log which is much better than the crash.

kbingham, 2022-11-17:
This https://patchwork.libcamera.org/patch/17290/ was merged in September, which I think should also fix / prevent the issue you hit here. If you're able to retest this - let me know if it's still an issue or we can close this off.