kbingham/libcamera #37: sample code for using libcamera

erasta, 2022-02-15:
Hello,
Is there a sample code of how to use `libcamera`? or can you supply one?
Just a basic main function that:
- start the camera. 
- capture one photo into pixel memory array. 
- close the camera. 

`libcamera-apps` code is really good and should do that, but it has a complicated pipeline for showing the preview or handling various output modes.
That would greatly help.
Many thanks for this great repo.

kbingham, 2022-02-15:
https://git.libcamera.org/libcamera/simple-cam.git/

erasta, 2022-02-15:
awesome! thanks!
adding its github link for future use:
https://github.com/kbingham/simple-cam

closed 2022-02-15 by erasta

kbingham, 2022-02-15:
Mapping the buffer with mmap is a bit of a pain, that internally in libcamera we have a helper class called MappedFrameBuffer.

It's not exposed by the public api, as other developers don't want that to be part of the interfaces for libcamera - but I you are welcome to duplicate that class and use it in your application.

Somehow I want to try to make that class explicitly in the public domain until there's a better "Image" class that has been proposed.