kbingham/libcamera #38: Text / timestamp overlay

webgps, 2022-02-23:
Please add support for text / timestamp overlay on picture.

kbingham, 2022-02-23:
This seems quite out of scope for libcamera itself.

Are you looking for a specific functionality or feature requirement? What application are you using to capture, where you want the text/timestamp to be added?

kbingham, 2022-03-16:
@webgps If you want to add a timestamp overlay you could try something like the following example using gstreamer:

```
gst-launch-1.0 libcamerasrc !  \
       video/x-raw,colorimetry=bt709,format=NV12,width=1280,height=720,framerate=30/1 ! \
       clockoverlay ! jpegenc ! multipartmux ! \
       tcpserversink host=0.0.0.0 port=5000
```

That will capture, add a timestamp, and encode as MJPEG. You can then view it on another remote device with:

```
gst-launch-1.0 tcpclientsrc host=$DEVICE_IP port=5000 ! \
        multipartdemux ! jpegdec ! autovideosink
```

I hope this helps your testing/use case. If it answers your question, please close the issue.

webgps, 2022-03-17:
> @webgps If you want to add a timestamp overlay you could try something like the following example using gstreamer:
>
> ```
> gst-launch-1.0 libcamerasrc !  \ [...]

Perfect! Thank you very much, just what i wanted.
Only thing, that I don't know why but the stream isn't seem like a 30fps stream, more like a 5fps. I'm using a 16mp autofocus (imx519) arducam. Do you know why is that?

kbingham, 2022-04-11:
those properties are not yet developed in the gstlibcamerasrc - patches are welcome to add them! We have a current GSoC student working on extending the gstreamer elements too - so there should be improvements after the summer (or along the way) - and I'm sure asking on the libcamera mailing list - or IRC channel will get some support if you need it too.