kbingham/libcamera #41: Hitting build issue on Raspi 4 (pymain.cpp)

kevleyski, 2022-05-12:
Not having great success building on pi4 (2GB) - was following whats written [here](https://libcamera.org/getting-started.html)

[260/415] Compiling C++ object src/py/libcamera/_libcamera.so.p/pymain.cpp.o
FAILED: src/py/libcamera/_libcamera.so.p/pymain.cpp.o
c++ -Isrc/py/libcamera/_libcamera.so.p -Isrc/py/libcamera -I../src/py/libcamera -Iinclude -I../include -I../subprojects/pybind11/include -Iinclude/libcamera -I/usr/include/python3.9 -I/usr/include/aarch64-linux-gnu/python3.9 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Werror -std=c++17 -O0 -g -Wshadow -include config.h -fPIC -fvisibility=hidden -Wno-shadow -DPYBIND11_USE_SMART_HOLDER_AS_DEFAULT -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/py/libcamera/_libcamera.so.p/pymain.cpp.o -MF src/py/libcamera/_libcamera.so.p/pymain.cpp.o.d -o src/py/libcamera/_libcamera.so.p/pymain.cpp.o -c ../src/py/libcamera/pymain.cpp
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.

kevleyski, 2022-05-13:
lol I think its just /usr/lib/gcc/aarch64-linux-gnu/10/cc1plus maxing out the raspi4 unit

I'll cross compile instead :-)

kbingham, 2022-05-13:
Yes, parallel builds don't work on the RPi4 with lower memory - as it quickly exhausts all available RAM.

I've started : https://gist.github.com/kbingham/b35cc1402d02b0ef41b450941ad8f0c3 recently as a way I hope can simplify cross compiling for RPi - but it's a work in progress, I've compiled, but not yet worked out how to packaage/install the results into a device. Might help you though.

kbingham, 2022-05-13:
If you try it, please let me know how you get on - and how we might improve it - then it could be incorporated into the project somehow perhaps…

kevleyski, 2022-05-19:
Yeah that looks good (though I'm not using bullseye so my baseline docker is different, but similar)

kbingham, 2022-05-19:
Also - you can try 'ninja -j1' to make sure you only do a single thread build on the RPi - it's really slow - but it might have a better chance of surviving a compile without exhausting the RAM.

kevleyski, 2022-05-19:
Yeah had been doing that too, works just takes a bit longer :-)

kbingham, 2022-07-11:
My original simple docker gist has evolved a little into this, and I still hope to further get this into something that coudl be then later merged to libcamera itself too.

https://github.com/CactiChameleon9/Libcamera-RPiOS-Build-Enviroment

closed 2022-07-11 by kbingham