kbingham/libcamera #42: Need a Stable branch or Tagged release to integrate libcamera.

savidha777, 2022-07-11:
Hi,

We are trying to integrate below libcamera source code for RPI platform.
Libcamera source code : **git://linuxtv.org/libcamera.git** 

Can you please share any stable branch or tagged release for our integration.

Thanks,

kbingham, 2022-07-11:
Could you explain more about your requirements please?

What do you need  beyond a specific tag of the latest point. RPi already integrate libcamera into their distribution. Are you building a different root filesystem or OS?

savidha777, 2022-07-11:
We are working for RDK-Camera profile and using RPI platform for RDK-Camera use case validation.

We are willing to integrate libcamera to our RDK-C profile for to use libcamera providing features.

Need to compiled libcamera in RDK-Camera build, so we need to know which branch of code we have to refer for the integration.

kbingham, 2022-07-11:
Will you be at the RDK conference tomorrow / Wednesday? Please look out for me and find me there.

savidha777, 2022-07-11:
I am not attending the Tech Summit,  Please provide your feedback when you have time. Thanks !!!

kbingham, 2022-07-11:
Ahem, not sure how to mark this as a duplicate - but I'll keep #5 related to this already.

closed 2022-07-11 by kbingham

kbingham, 2022-10-14:
https://git.libcamera.org/libcamera/libcamera.git/tag/?h=v0.0.1

Also - libcamera is packaged in openembedded, so should be easy to use with yocto.

http://cgit.openembedded.org/meta-openembedded/tree/meta-multimedia/recipes-multimedia/libcamera/libcamera.bb?h=master