kbingham/libcamera #44: libcamera support for single camera with multiple streams

Sajnakn, 2022-07-27:
Does libcamerawe have support for single camera source with 4 hardware encoder output?
A single device node represents all these four streams. A stream id provided in the ioctl differentiates the streams.

kbingham, 2022-07-27:
Can you clarify which hardware you are referencing please?

libcamera does indeed support multiple streams from a camera, but it is dependant upon the configuration of the ISP hardware.

Sajnakn, 2022-07-28:
It is SOC Ambrella S3L camera. It has 4 different streams

kbingham, 2022-07-28:
Are the streams generated by the ISP or a separate encoding block? libcamera does not impose a restirction on the number of streams a pipeline handler can create or expose for the device - but I suspect it would be very useful to go through the actual architecture design of your pipeline handler.

kbingham, 2022-07-28:
Please e-mail me on kieran.bingham@ideasonboard.com - I think there is quite a lot more context required to discuss to determine your requirements.

closed 2023-07-05 by kbingham