kbingham/libcamera #45: gcc version requirement for libcamera

Sajnakn, 2022-07-27:
While compiling libcamera, I got errors like:
	../include/libcamera/base/bound_method.h:210:64: error: ‘maybe_unused’ attribute directive ignored [-Werror=attributes]

And solved by upgrading the gcc version to 9. 
Is the gcc version above 8 is mandatory to build libcamera? Or can we compile it with gcc version 5?

kbingham, 2022-07-27:
GCC 8 should indeed be supported, though I see in my test matrix, I'm only compiling with GCC 9 and above, and clang 11 and above, however we have a buildbot that compiles for gcc-8.
https://buildbot.libcamera.org/#/builders/6

Can you paste the output of gcc-8 --version here please?

Sajnakn, 2022-07-28:
`$ gcc-8 --version
gcc-8 (Ubuntu 8.4.0-1ubuntu1~16.04.1) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

Sajnakn, 2022-07-28:
Can we compile at least the base of the libcamera with gcc5, without compiling the optional module?

kbingham, 2022-07-28:
I think GCC5 is very old. (Too old) Are you sure you mean gcc5 ? Is that really what you need to compile libcamera with?

Sajnakn, 2022-07-28:
The gcc version we use to compile the complete stack is gcc5.2 for arm(linaro-armv7ahf-2015.11-gcc5.2) only. I am trying to integrate libcamera to the stack.

kbingham, 2022-07-28:
I'm afraid you might struggle with libcamera then. We use C++17 internally for instance, so a compiler from 2015 is likely going to struggle.

What's preventing you from using a newer toolchain?

Sajnakn, 2022-08-04:
Current gcc version we are provided by vendors is gcc 5. In the scripts it is mentioned that gcc 8 or more is required for compiling libcamera. But with gcc-8 also, we got some errors and after upgrading to gcc9 all those errors resolved. So, which one you recommend?

kbingham, 2022-08-04:
We support gcc-8 and above. There is a known bug compiling on early versions of gcc-8 which I believe is being worked on already.

closed 2023-07-04 by kbingham