kbingham/libcamera #46: Authentication error with ninja -C build install

JHPHELAN, 2022-08-23:
`Following from:
https://libcamera.org/getting-started.html
and
https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-libcamera-apps
Skip over Getting Started to install dependencies first
Check for C++
ubuntu@ubuntu:~$ g++ --version
g++ (Ubuntu 11.2.0-19ubuntu1) 11.2.0

ubuntu@ubuntu:~$ pip3 install --user meson
ubuntu@ubuntu:~$ pip3 install --user --upgrade meson

ubuntu@ubuntu:~/libcamera$ pip3 install ninja

Being unclear as to which of the optional dependencies may be needed
install then all.  Respond “y” when asked.
ubuntu@ubuntu:~$ sudo apt install libyaml-dev python3-yaml python3-ply python3-jinja2
ubuntu@ubuntu:~/libcamera$ sudo apt install libgnutls28-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install libdw-dev libunwind-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install libudev-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install python3-sphinx doxygen graphviz texlive-latex-extra
ubuntu@ubuntu:~/libcamera$ sudo apt install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install libevent-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install libdrm-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install libjpeg-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install libsdl2-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5 qttools5-dev-tools libtiff-dev
ubuntu@ubuntu:~/libcamera$ sudo apt install liblttng-ust-dev python3-jinja2 lttng-tools

Configure /boot/firmware/config.txt
ubuntu@ubuntu:~$ cd /boot/firmware/
ubuntu@ubuntu:/boot/firmware$ sudo nano config.txt
	camera_auto_detect=0		[change existing line =1 > =0]
	dtoverlay=imx219		[add new line at bottom for Picamera driver]
	start_x=1			[may also be needed at end to enable camera]

Now to the Getting Started part:
ubuntu@ubuntu:~$ git clone https://git.libcamera.org/libcamera/libcamera.git
ubuntu@ubuntu:~$ cd libcamera
ubuntu@ubuntu:~/libcamera$ meson build
ubuntu@ubuntu:~/libcamera$ ninja -C build install
very long install...then -
[424/425] Installing files.
Installation failed due to insufficient permissions.
Attempting to use polkit to gain elevated privileges...
==== AUTHENTICATING FOR org.freedesktop.policykit.exec ===
Authentication is needed to run `/usr/bin/python3' as the super user
Authenticating as: Ubuntu (ubuntu)
Password:
polkit-agent-helper-1: error response to PolicyKit daemon: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: No session for cookie
==== AUTHENTICATION FAILED ===
Error executing command as another user: Not authorized

This incident has been reported.
FAILED: meson-internal__install
/home/ubuntu/.local/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.`

What's going on here and how to fix it?
Thanks!
Jim Phelan
PS pardon the weird formatting.  Don't know why it's making strikeouts.

eli-schwartz, 2022-08-23:
You are trying to install libcamera with the default installation paths i.e. installation to `/usr/local` -- you do not have permission to do this as your current user, so Meson attempted to re-run itself as root, using pkexec.

This failed as well (did you not have a root password? Did you not type it in correctly? Either way, this is a matter of polkit permissions, not meson and not libcamera specifically). Since that failed, `meson install` returned an overall error -- it could not find a way to install to /usr/local.

> PS pardon the weird formatting. Don't know why it's making strikeouts.

Try making the entire console paste go inside triple backticks

~~~
```
text goes here
```
~~~

JHPHELAN, 2022-08-23:
When the request came up for password I did enter the correct root password.
So is "Getting Started" assuming I'm acting as root?  Should I then to "sudo meson build" and "sudo ninja -C build install"?

kbingham, 2022-08-23:
Actually I think it's /your/ password not the root password, and your account should have permission to have sudo access I expect.
the prints " ==== AUTHENTICATION FAILED ===" mean that indeed - something went wrong either with the password, or the account not having permission.

Indeed this is a meson/ninja issue - not a libcamera issue.

eli-schwartz, 2022-08-23:
polkit can be configured all different ways, e.g. you can choose to auth as the current user, or as the root user, or choose a random other user with privileges and auth as *that* user.

At the end of the day, it very much boils down to polkit authentication failing. If you prefer to use sudo instead of polkit you can of course do that too :) `sudo meson install -C build/ --no-rebuild`

JHPHELAN, 2022-08-23:
I'm logged in as the default ubuntu user and am a member of the sudo group

eli-schwartz, 2022-08-23:
polkit may or may not be configured to care about the sudo group. Either way, the point is that this will work if and only if polkit (not sudo) is successfully authenticated.

As I said above, you can run the installation with sudo yourself rather than having Meson internally re-run itself with the authentication mechanism that doesn't work for you.

closed 2024-11-29 by kbingham