kbingham/libcamera #49: ERROR Serializer control_serializer.cpp:511 Bad data, entry offset….

ToivaMa, 2022-09-02:
I have been installed these:
- RasPi4 new installation.
- Nodejs
- GIT
- meson
- python3
- Ninja
- libcamera
After this camera works Ok.

Then i make own cert by mkcert. And then get this message:
raspcamera1:~ $ libcamera-hello
Preview window unavailable
[21:35:00.098312980] [4146]  WARN IPAManager ipa_manager.cpp:113 Public key not valid
[21:35:00.102368120] [4146]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3855-6c6289ee-dirty (2022-09-02T08:24:44+03:00)
[21:35:00.176954143] [4147] ERROR Serializer control_serializer.cpp:511 Bad data, entry offset mismatch (entry 2), entry->offset = 32, value.offset() = 36
[21:35:00.180670683] [4147]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media2
libcamera-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera_app.so: undefined symbol: _ZNK9libcamera11ControlList8containsERKNS_9ControlIdE

kbingham, 2022-09-02:
Hello, Can I ask why you're making your own certificates?
The error message shows that the certificates are not valid (Public key not valid) which is then running the IPA component as an isolated module. That seems to be failing, so it's certainly a bug - but I wouldn't expect you to need to make your own certificates to sign the IPA.

kbingham, 2022-09-02:
The error:
 libcamera-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera_app.so: undefined symbol: _ZNK9libcamera11ControlList8containsERKNS_9ControlIdE 

Makes me wonder if there is a mismatch between the compiled libcamera, and libcamera-apps too. Did you self compile anything?

ToivaMa, 2022-09-06:
Hello @kbingham,
I made a certificates for the Nodejs web application. Nothing was compiled.
I think i do new fresh install without mkcert.

kbingham, 2022-09-06:
In fact I think we have a good for this one already

kbingham, 2022-09-06:
I think this is the same issue as being solved by this patch.

https://patchwork.libcamera.org/patch/17292/

kbingham, 2022-09-09:
Another solution for this is to install openssl locally. That should fix (workaround) the 
` [21:35:00.098312980] [4146] WARN IPAManager ipa_manager.cpp:113 Public key not valid `
error print, and the secondary failure then won't occur.

That's definitely all due to changes in libcamera - so we'll get these resolved as soon as we can.

kbingham, 2022-10-17:
Have you tried the latest libcamera branches ? I think these issues should be resolved. If not can you let me know please?