kbingham/libcamera #50: Zero sequence expected for first frame error on Rpi4

tmsd2001, 2022-09-02:
OS:
Linux camerapi8 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

libcamera:
```

[1:17:14.018313530] [15880]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3865-3fad116f
[1:17:14.032367040] [15881]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media2
[1:17:14.033062467] [15880]  INFO Camera camera.cpp:1035 configuring streams: (0) 1296x972-YUV420
[1:17:14.033441180] [15881]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
```
Error:
```
Stream configuration adjusted
[0:34:20.004970168] [13509]  INFO Camera camera.cpp:1035 configuring streams: (0) 2592x1944-YUV420 (1) 2592x1944-SGRBG10_CSI2P
[0:34:20.005919105] [13510]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGRBG10_1X10 - Selected unicam format: 2592x1944-pgAA
TfStage: no low resolution stream
TfStage: Main stream is 2592x1944
 detected
[0:34:20.250307291] [13510]  WARN V4L2 v4l2_videodevice.cpp:1803 /dev/video0[15:cap]: Zero sequence expected for first frame (got 1)
```
App:
libcamera-apps / libcamera-detect
The program gets stuck at this point, but hasn't crashed.

kbingham, 2022-09-02:
That last warning about zero sequence can be ignored. Perhaps we should find a way to fix it in the kernels so it doesn't appear, but it's not harmful in this instance. In fact Perhaps it could even be lowered to a debug print.

As to being stuck, If you are using libcamera-apps, please report this issue to them.
Raspberry Pi develop the libcamera-apps independently of the libcamera project.

   https://github.com/raspberrypi/libcamera-apps

kbingham, 2022-10-17:
Are there still any topics to consider in this issue? 

I've prepared and sent a patch to lower the log level of this print to hopefully reduce concern about it in the future.
https://patchwork.libcamera.org/patch/17611/

trieska, 2023-12-17:
well, this is bit old thread so maybe I should look also somewhere else, but trying to find answers

I have python program which is sending image over mqtt but also can save image to disk
I am using pizerow with original picamera, did install `python3-picamera2`
I did check, everything is updated

shortly my program doing these steps

     from picamera2 import Picamera2

object init, as you can see that I am saving mqtt config to separate variable to quick used later

    self.picam2 = Picamera2()
    self.mqtt_config = self.picam2.create_still_configuration(main={"size": (800, 600)})
    self.picam2.configure(self.picam2.create_still_configuration())

here is saving method, this one is working fine

    def save_photo(self):
        file_name = self.create_file_name()
        self.picam2.capture_file(file_name)

and here is send method, but time to time this one hang on second line "switch_mode_and_capture"

    def send(self, client):
        data = io.BytesIO()
        self.picam2.switch_mode_and_capture_file(self.mqtt_config, data, format='jpeg')
        client.publish(self.getTopic(), data.getvalue())
        data.close()
    

here is output from system log when is hang on

      INFO Camera camera.cpp:1033 configuring streams: (0) 800x600-BGR888 (1) 1296x972-SGBRG10_CSI2P
      INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
      INFO V4L2 v4l2_videodevice.cpp:1820 /dev/video0[20:cap]: Zero sequence expected for first frame (got 1)

othewise looks like this:

    INFO Camera camera.cpp:1033 configuring streams: (0) 800x600-BGR888 (1) 1296x972-SGBRG10_CSI2P
    INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
    INFO Camera camera.cpp:1033 configuring streams: (0) 2592x1944-BGR888 (1) 2592x1944-SGBRG10_CSI2P
    INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA

what do you think? what could be problem? should I change my program?
thanks

kbingham, 2023-12-17:
If you're having issues with picamera2, please report it there.
https://github.com//raspberrypi/picamera2

closed 2023-12-17 by kbingham