kbingham/libcamera #52: Cam utility is not listing the camera in RPI 3B+

savidha777, 2022-09-19:
Hi

In RDK-C RPI camera image, Added "dtoverlay=imx219" in /boot/config.txt file and rebooted the system.

Once system rebooted , tried to list out available camera in RPI system with **'cam -l'** utility.

But it is not listing the camera. After the some prints binary is hanged.

**Used latest libcamera revision : 8b02645845bf404fba83910818b285b652f14f89 (20Jul2022)**

Please find below log:
root@raspberrypi3-rdk-camera:~# LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l
[0:15:04.685612102] [3456] DEBUG IPAModule ipa_module.cpp:329 ipa_ipu3.so: IPA module /usr/lib/libcamera/ipa_ipu3.so is signed
[0:15:04.686569081] [3456] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/libcamera/ipa_ipu3.so'
[0:15:04.687005384] [3456] DEBUG IPAModule ipa_module.cpp:329 ipa_rkisp1.so: IPA module /usr/lib/libcamera/ipa_rkisp1.so is signed
[0:15:04.687380175] [3456] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/libcamera/ipa_rkisp1.so'
[0:15:04.687923977] [3456] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/lib/libcamera/ipa_rpi.so is signed
[0:15:04.688536894] [3456] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/libcamera/ipa_rpi.so'
[0:15:04.688908821] [3456] DEBUG IPAModule ipa_module.cpp:329 ipa_vimc.so: IPA module /usr/lib/libcamera/ipa_vimc.so is signed
[0:15:04.689244290] [3456] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/libcamera/ipa_vimc.so'
[0:15:04.689602311] [3456]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:15:04.690210644] [3457] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[0:15:04.704325644] [3457] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media2
[0:15:04.704540644] [3457] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media2 due to 3 missing dependencies
[0:15:04.708609342] [3457] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media2 found
[0:15:04.708753040] [3457] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: bcm2835-codec

Binary is hanged after this print.

#################################################################################################

But able to list out camera in RPI system with oldest libcamera revision

**Used libcamera revision :  58d487b10e06ad823103957c1e6aa2201720fdae(30Oct2020)**

Working log:
[Working_old_libcamera_log.txt](https://github.com/kbingham/libcamera/files/9599472/Working_old_libcamera_log.txt)

Any dependency or configurations are needed to list out camera in latest libcamera source code.

kbingham, 2022-09-19:
Are you running an up to date kernel and boot loader firmware (from RPi) as well?

kbingham, 2022-09-19:
> Used latest libcamera revision : https://github.com/kbingham/libcamera/commit/8b02645845bf404fba83910818b285b652f14f89 (20Jul2022)

This isn't the latest libcamera revision. The master branch currently has 74ab3f778c848b20cbf8fe299170756ff6ebab1a is already a fair few commits on. It's worth checking the latest master which has some updates from RPi for camera stability - but I don't think in this instance it will have much effect.

savidha777, 2022-09-20:
Observing same hang issue with this latest revision 74ab3f778c848b20cbf8fe299170756ff6ebab1a.

Currently we are using below kernel and boot loader version for RPI

**Kernel:**
SRC URI : git://github.com/raspberrypi/linux.git 
Branch : rpi-5.4.y

**Boot loader:**
https://github.com/raspberrypi/firmware/archive/1eb8921ae7158249894f89e8bb6c4bfb04ec7377.tar.gz
SRC_REV : 1eb8921ae7158249894f89e8bb6c4bfb04ec7377 (19Nov2020)

Requesting to share libcamera required RPI kernel and boot loader version.

kbingham, 2022-09-20:
> Requesting to share libcamera required RPI kernel and boot loader version.

That's really a question for Raspberry Pi ... and perhaps should be asked on their forums.

Alternatively, could you move this issue to https://github.com/raspberrypi/libcamera please?

savidha777, 2022-09-20:
Okay

savidha777, 2022-09-20:
Is this hang issue is really based on RPI kernel and boot loader version or some thing else creating the issue?

any idea on this...

kbingham, 2022-09-20:
I don't know yet. It sounds very Raspberry Pi specific to me ... hence lets move it to RPi.

kbingham, 2022-09-20:
Closing here, we can reopen if it's not RPi specific.

closed 2022-09-20 by kbingham

savidha777, 2022-11-03:
Hi,

We are observing same hang issue with upgraded rpi 5.15 kernel as well.

please find below used libcamera revision and rpi kernel revision.

libcamera : 685da13d02c01ed0a2bb7080f95b7ac56e80f221 (21Sep22)

RPI Kernel : rpi-5.15.y 
	     version - 5.15.74
	     revision - dbd073e4028580a09b6ee507e0c137441cb52650 (Sep22)
	     

Based on the analysis, Observing the hang issue in below API calling

**device_enumerator.cpp
	addDevice() -> devicesAdded.emit();**
	
Please find this attached break point output 
[04Nov22_cam_utility_HangIssue.txt](https://github.com/kbingham/libcamera/files/9932276/04Nov22_cam_utility_HangIssue.txt)

Thanks!!!

kbingham, 2022-11-04:
Can you check the bt of all threads?

```
info threads
thread apply all bt
```

kbingham, 2022-11-04:
Please make sure you have all logging enabled too.

set 
`LIBCAMERA_LOG_LEVELS=*:0 cam -l`

kbingham, 2022-11-04:
Did you open an issue for this at  https://github.com/raspberrypi/libcamera ?