kbingham/libcamera #54: Build libcamera against Android SDK

kbingham, 2022-10-14:
References to get started:

https://proandroiddev.com/how-to-setup-android-sdk-without-android-studio-6d60d0f2812a?gi=6f859505330e 

https://github.com/ppetraki/meson-android-helloworld

rsglobal, 2022-10-14:
You probably meant NDK: https://developer.android.com/ndk/downloads

As far as I know SDK is for Android apps.

rsglobal, 2022-10-14:
Here's the starting point:

1. Download [ndk](https://developer.android.com/ndk/downloads) into /opt/ndk directory
2. Clone libcamera
3. Download build.sh and meson_cross_compile.txt into libcamera directory
4. Run ./build.sh

[meson_cross_compile.txt](https://github.com/kbingham/libcamera/files/9785767/meson_cross_compile.txt)
[build.zip](https://github.com/kbingham/libcamera/files/9785770/build.zip)

rsglobal, 2022-10-16:
There're some system dependencies to be installed before building (especially when cross-compiling for ARM). It would be nice to came-up with a dockerfile which will include all required dependencies.

kbingham, 2022-10-17:
Is https://github.com/CactiChameleon9/Libcamera-RPiOS-Build-Enviroment useful to you ?

kbingham, 2022-10-17:
Which part of your solution is not upstreamable? The other participant of the other issue is working on the mailing list to determine adding (I expect /your/ makefile wrapper).

The work can only get upstream if someone works on it.

rsglobal, 2022-10-17:
> Which part of your solution is not upstreamable?

Here's my WIP branch. It's very dirty right now, but it can partially answer the question.
https://github.com/GloDroid/glodroid_forks/commits/libcamera-next

> The work can only get upstream if someone works on it.

That won't be me, sorry. I'm already supporting a lot of FOSS repos. And only benefit I have from my FOSS activity is a feeling that world is becoming better. Nobody want to pay for that.

rsglobal, 2022-11-15:
@kbingham ,

Can you help me please to understand this line?
```
adb shell LIBCAMERA_LOG_LEVELS=*:DEBUG lc-compliance
[17:41:46.203161361] [5713]  WARN IPAManager ipa_manager.cpp:114 Public key not valid
```

Why public key may be invalid?

Here's my generated key: 
[ipa_pub_key.cpp](https://github.com/kbingham/libcamera/files/10013048/ipa_pub_key.cpp.txt)

kbingham, 2022-11-15:
I've tried following these steps to build with android ndk:

I've created a meson cross file as mesoncross/android-ndk-r25b
```
[binaries]
ar = '/opt/ndk/android-ndk-r25b/toolchains/llvm/prebuilt/linux-x86_64/bin/llvm-ar'
c = '/opt/ndk/android-ndk-r25b/toolchains/llvm/prebuilt/linux-x86_64/bin/x86_64-linux-android33-clang'
cpp = '/opt/ndk/android-ndk-r25b/toolchains/llvm/prebuilt/linux-x86_64/bin/x86_64-linux-android33-clang++'
c_ld = 'lld'
cpp_ld = 'lld'
pkgconfig = '/usr/bin/pkg-config'
strip = '/opt/ndk/android-ndk-r25b/toolchains/llvm/prebuilt/linux-x86_64/bin/llvm-strip'

[host_machine]
system = 'linux'
cpu_family = 'x86_64'
cpu = ''
endian = 'little'

```
and configured with:

```
meson build/android-ndk-25b/ \
	--cross mesoncross/android-ndk-r25b \
	-Dsysconfdir=/vendor/etc \
	-Dandroid=enabled

...

libcamera 0.0.1

  Versions
    Sources                  : 0.0.1+70-0eb4ccd3

  Paths
    LIBCAMERA_DATA_DIR       : "/usr/local/share/libcamera"
    LIBCAMERA_SYSCONF_DIR    : "/vendor/etc/libcamera"
    IPA_PROXY_DIR            : "/usr/local/libexec/libcamera"
    IPA_CONFIG_DIR           : "/vendor/etc/libcamera/ipa:/usr/local/share/libcamera/ipa"
    IPA_MODULE_DIR           : "/usr/local/lib/libcamera"

  Configuration
    Enabled pipelines        : ipu3
                               raspberrypi
                               rkisp1
                               simple
                               uvcvideo
                               vimc
    Enabled IPA modules      : ipu3
                               raspberrypi
                               rkisp1
                               vimc
    Tracing support          : YES
    Android support          : YES
    GStreamer support        : YES
    Python bindings          : NO
    V4L2 emulation support   : NO
    cam application          : YES
    qcam application         : YES
    lc-compliance application: YES
    Unit tests               : NO

  Subprojects
    libyuv                   : YES

  User defined options
    Cross files              : mesoncross/android-ndk-r25b
    sysconfdir               : /vendor/etc
    android                  : enabled

```

But I get a lot of host header pollution:

```
In file included from /opt/ndk/android-ndk-r25b/toolchains/llvm/prebuilt/linux-x86_64/bin/../sysroot/usr/include/features.h:36:
/usr/include/x86_64-linux-gnu/sys/cdefs.h:64:6: error: function-like macro '__GNUC_PREREQ' is not defined
# if __GNUC_PREREQ (4, 6) && !defined _LIBC
     ^
```

Do you get this? or have you already resolved such issues ?

kbingham, 2022-11-15:
```
pkgconfig = '/usr/bin/pkg-config'
```
Seems 'suspicious' to me...

rsglobal, 2022-11-15:
Try `pkgconfig = ['env', 'PKG_CONFIG_LIBDIR='path_to_NDK_pkgs_dir', '/usr/bin/pkg-config']`

rsglobal, 2022-11-15:
I haven't tried to build libcamera using NDK yet. I think it would require some minimal sysroot preparation with packages and C headers from AOSP.

kbingham, 2022-11-15:
Oh - I thought https://github.com/kbingham/libcamera/issues/54#issuecomment-1278894473 was something you had done.
I don't seem to have much in the way of a sysroot for the NDK indeed, so not much for the pkg-config to point to ..

rsglobal, 2022-11-15:
May be useful:

That's how env. variables can be overridden for service to enable logcat debug:

```diff
+++ /vendor/etc/init/android.hardware.camera.provider@2.5-service_64.rc
service vendor.camera-provider-2-5 /vendor/bin/hw/android.hardware.camera.provider@2.5-service_64
    interface android.hardware.camera.provider@2.5::ICameraProvider legacy/0
    interface android.hardware.camera.provider@2.4::ICameraProvider legacy/0
    class hal
    user cameraserver
    group audio camera input drmrpc
    ioprio rt 4
    capabilities SYS_NICE
    task_profiles CameraServiceCapacity MaxPerformance
+    setenv LIBCAMERA_LOG_LEVELS *:DEBUG
+    setenv LIBCAMERA_LOG_FILE syslog
```

rsglobal, 2022-11-15:
> Oh - I thought https://github.com/kbingham/libcamera/issues/54#issuecomment-1278894473 was something you had done.

Not yet, but have the plans to do it in a future.

kbingham, 2022-11-15:
Ok, good to know you were able to build ... I got further with setting the sysroot - but still get linker errors:
```
ld.lld: error: unable to find library -ldw
ld.lld: error: unable to find library -lelf
```
So I'll have to do some more digging next time I get chance.

kbingham, 2022-11-15:
So preventing finding the host libraries requires setting (**correctly, without typos**!) the `pkg_config_libdir` property in the mesoncross file.

But now I'm hit by not having libexif available in the NDK sysroot, which is required for the android layer. Next steps, identify how to provide libexif to the NDK sysroot.