kbingham/libcamera #55: Capturing to avi format with libav codec writes wrong fps to file

ShovelJockey, 2022-10-15:
Hi there I am aiming to record 1 hr videos at 500x375) from a raspberry pi (running 64-bit bullseye).

Using this command line in a bash script:

`libcamera-vid -t $filmDuration --framerate 5 --width 500 --height 375 --nopreview -- --codec libav --libav-format avi -o "$(date +%Y%m%d_%H%M.avi)" --tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json` 

This records the footage as needed but the file information incorrectly has 600.000fps which prevents use of this video with other software. 

This figure is not accurate apparent both from watching, them having the correct number of frames for 3600 seconds at 5 fps and software that interprets the 600fps as true quickly running out of frames.

Am I missing something here or is this unintended behaviour? I found that by changing the container to mp4 the information was correct but any interruption of the recording would corrupt the video missing moov files.

Any help would be greatly appreciated.

kbingham, 2022-10-17:
If you're using libcamera-vid - this will need to be raised on the Raspberry Pi camera forums. Whilst 'libcamera-vid' shares the name 'libcamera' - it's not written or maintained by the libcamera project, and is entirely supported by Raspberry Pi.

If you want to replicate this pipeline with gstreamer, and hit any issues, then that would be something the libcamera project can support.

ShovelJockey, 2022-10-18:
Ok thanks will close this issue.

closed 2022-10-18 by ShovelJockey