kbingham/libcamera #59: Build error on clang with -Dtest=true

rsglobal, 2022-11-14:
```
../src/test/log/log_process.cpp:102:52: error: 'open' has superfluous mode bits; missing O_CREAT? [-Werror,-Wuser-defined-warnings]
                int fd = open(logPath_.c_str(), O_RDONLY, S_IRUSR);
                                                                 ^
```

rsglobal, 2022-11-14:
Another one:
```
In file included from ../src/src/apps/lc-compliance/simple_capture.cpp:8:
/usr/src/gtest/include/gtest/gtest.h:1358:11: error: comparison of integers of different signs: 'const int' and 'const unsigned int' [-Werror,-Wsign-compare]
  if (lhs == rhs) {
      ~~~ ^  ~~~
/usr/src/gtest/include/gtest/gtest.h:1377:12: note: in instantiation of function template specialization 'testing::internal::CmpHelperEQ<int, unsigned int>' requested here
    return CmpHelperEQ(lhs_expression, rhs_expression, lhs, rhs);
           ^
../src/src/apps/lc-compliance/simple_capture.cpp:51:2: note: in instantiation of function template specialization 'testing::internal::EqHelper::Compare<int, unsigned int, nullptr>' requested here
        EXPECT_EQ(count, config_->at(0).bufferCount) << "Allocated less buffers than expected";
        ^
/usr/src/gtest/include/gtest/gtest.h:1848:54: note: expanded from macro 'EXPECT_EQ'
  EXPECT_PRED_FORMAT2(::testing::internal::EqHelper::Compare, val1, val2)
```

kbingham, 2022-11-14:
Can you confirm the version of your compiler and which version of libcamera you are building please?

rsglobal, 2022-11-14:
Latest libcamera master / clang version 14.0.6

kbingham, 2022-11-14:
Hrm, I definitely build with clang 14 on the integration tests. Can you capture a full clean build log, and the exact version of the compiler please? I'll take a look at the build failures anyway, but it would help to be able to replicate it.

rsglobal, 2022-11-15:
I'm building within AOSP, clang --version output is:

```
Android (8490178, based on r450784d) clang version 14.0.6 (https://android.googlesource.com/toolchain/llvm-project 4c603efb0cca074e9238af8b4106c30add4418f6)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /home/roman/glodroid/prebuilts/clang/host/linux-x86/clang-r450784d/bin
```

rsglobal, 2022-11-15:
Google may have enabled more experimental checks, which isn't enabled on classic clang14 release. You may try clang15.

rsglobal, 2022-11-15:
Since every new compiler version brings more checks, build will likely to fail with them.
What about dropping -Werror from meson.build ?
I would enforced it only in CI to catch errors at patch application stage.

kbingham, 2022-11-15:
When you configure, I believe you can disable this flag yourself (as you are the distribution).

 meson configure -Dwerror=false

rsglobal, 2022-11-15:
I'll try that, thanks

kbingham, 2022-11-15:
> Since every new compiler version brings more checks, build will likely to fail with them.

Honestly, we /want/ to know about this as early as possible. That's why we keep -Werror enabled. But I certainly understand you don't want to fix them all.

closed 2024-11-29 by kbingham