kbingham/libcamera #60: Got lc-compliance crash on rpi4

rsglobal, 2022-11-15:
```
Searching for native crashes in tombstone_41
Reading symbols from /home/roman/glodroid/out/target/product/rpi4/symbols
Revision: '1.0'
pid: 4584, tid: 4584, name: lc-compliance  >>> lc-compliance <<<
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0000000000000028
    x0  b4000077eab80338  x1  b40000779ab7e130  x2  b40000776ab7a870  x3  0000007fdfe772f8
    x4  0000000000000030  x5  000000775ab74a00  x6  0000083100000876  x7  0000083d00000843
    x8  b4000077eab802d0  x9  0000000000000000  x10 0000000000000040  x11 0000000000000011
    x12 000000796ac4f680  x13 0000000000000009  x14 000000000000086a  x15 0000000000000000
    x16 000000797018e7a0  x17 0000007970140710  x18 00000079738b0000  x19 b4000077dab7ea80
    x20 b40000779ab7e130  x21 b40000777ab805b0  x22 0000000000000002  x23 b40000776ab7ac40
    x24 00000000ffffffff  x25 0000005e1a8abe60  x26 0000005e1a8ac0c6  x27 0000000000000000
    x28 b40000776ab7ac40  x29 0000007fdfe772f0
  sp  0000007fdfe772f0  pc  0000007970140748
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory
WARNING: Cannot find /vendor/bin/lc-compliance in symbol directory

Stack Trace:
  RELADDR           FUNCTION                                                                                               FILE:LINE
  v-------------->  libcamera::FrameBuffer::Private::cancel()                                                              /home/roman/glodroid/out/target/product/rpi4/obj/LIBCAMERA/build/../src/include/l
  00000000000c2748  libcamera::Request::Private::doCancelRequest()+56                                                      /home/roman/glodroid/out/target/product/rpi4/obj/LIBCAMERA/build/../src/src/libca
  00000000000c2670  libcamera::Request::Private::~Private()+32                                                             /home/roman/glodroid/out/target/product/rpi4/obj/LIBCAMERA/build/../src/src/libca
  00000000000c2880  libcamera::Request::Private::~Private()+16                                                             /home/roman/glodroid/out/target/product/rpi4/obj/LIBCAMERA/build/../src/src/libca
  0000000000028324  SimpleCaptureBalanced::capture+1908                                                                    /vendor/bin/lc-compliance
  000000000002b2a8  SingleStream_Capture_Test::TestBody()+136                                                              /vendor/bin/lc-compliance
  0000000000066318  void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>+88                    /vendor/bin/lc-compliance
  0000000000049d14  testing::Test::Run()+660                                                                               /vendor/bin/lc-compliance
  000000000004aff4  testing::TestInfo::Run()+772                                                                           /vendor/bin/lc-compliance
  000000000004b888  testing::TestSuite::Run()+456                                                                          /vendor/bin/lc-compliance
  000000000005f184  testing::internal::UnitTestImpl::RunAllTests()+2468                                                    /vendor/bin/lc-compliance
  0000000000067008  bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>+88  /vendor/bin/lc-compliance
  000000000005e798  testing::UnitTest::Run()+120                                                                           /vendor/bin/lc-compliance
  000000000002471c  main+1660                                                                                              /vendor/bin/lc-compliance
  000000000004b630  __libc_init+96                                                                                         bionic/libc/bionic/libc_init_dynamic.cpp:157 (discriminator 2)
```

This diff workarounds the crash

```diff
--- a/src/libcamera/request.cpp
+++ b/src/libcamera/request.cpp
@@ -129,6 +129,8 @@ void Request::Private::doCancelRequest()
        Request *request = _o<Request>();
 
        for (FrameBuffer *buffer : pending_) {
+               if (buffer->_d() == nullptr)
+                       continue;
                buffer->_d()->cancel();
                camera_->bufferCompleted.emit(request, buffer);
        }
```

kbingham, 2022-11-15:
Yikes, that's well into the realm of "That shouldn't happen" (quite specifically, that doCancelRequest shouldn't be possible to have buffers attached that /don't/ have a valid _d() - so this workaround masks the underlying issue, but does highlight where we need to look.

I fear replicating it on android will be more difficult for me. But I'll run lc-compliance on an RPi4 now to make sure it's not happening there.

So I've just run lc-compliance on an RPi4 with an IMX219 - and got a different crash related to controls. It's an outdated device, so I'll rebuild the latest tree and retest, and then see if we need to pass this to RPi.

kbingham, 2022-11-15:
And now I believe I've reproduced it with a UVC webcam on my PC. I'll transfer this to a report on bugs.libcamera.org.

kbingham, 2022-11-15:
https://bugs.libcamera.org/show_bug.cgi?id=171