kbingham/libcamera #63: Crash when connecting certain devices

melvyn2, 2022-12-16:
Backtrace:
```c
systemd-coredump[54376]: Process 54367 (wireplumber) of user 1000 dumped core.
                                                   
                                                   Stack trace of thread 54374:
                                                   #0  0x00007fbd75ffbb1e _ZN9libcamera12ControlValueaSERKS0_ (libcamera.so.0.0.1 + 0x88b1e)
                                                   #1  0x00007fbd75ffd0c7 _ZN9libcamera11ControlInfoC1ENS_4SpanIKNS_12ControlValueELm18446744073709551615EEERS3_ (libcamera.so.0.0.1 + 0x8a0c7)
                                                   #2  0x00007fbd7602238e _ZN9libcamera10V4L2Device19v4l2MenuControlInfoERK19v4l2_query_ext_ctrl (libcamera.so.0.0.1 + 0xaf38e)
                                                   #3  0x00007fbd760226f5 _ZN9libcamera10V4L2Device15v4l2ControlInfoERK19v4l2_query_ext_ctrl (libcamera.so.0.0.1 + 0xaf6f5)
                                                   #4  0x00007fbd76022d00 _ZN9libcamera10V4L2Device12listControlsEv (libcamera.so.0.0.1 + 0xafd00)
                                                   #5  0x00007fbd7602307d _ZN9libcamera10V4L2Device5setFdENS_8UniqueFDE (libcamera.so.0.0.1 + 0xb007d)
                                                   #6  0x00007fbd76023125 _ZN9libcamera10V4L2Device4openEj (libcamera.so.0.0.1 + 0xb0125)
                                                   #7  0x00007fbd7603dbe2 _ZN9libcamera15V4L2VideoDevice4openEv (libcamera.so.0.0.1 + 0xcabe2)
                                                   #8  0x00007fbd7607a72c _ZN9libcamera13UVCCameraData4initEPNS_11MediaDeviceE (libcamera.so.0.0.1 + 0x10772c)
                                                   #9  0x00007fbd7607b3c8 _ZN9libcamera18PipelineHandlerUVC5matchEPNS_16DeviceEnumeratorE (libcamera.so.0.0.1 + 0x1083c8)
                                                   #10 0x00007fbd75ff6df1 _ZN9libcamera13CameraManager7Private22createPipelineHandlersEv (libcamera.so.0.0.1 + 0x83df1)
                                                   #11 0x00007fbd75ff70a7 _ZN9libcamera13CameraManager7Private4initEv (libcamera.so.0.0.1 + 0x840a7)
                                                   #12 0x00007fbd75ff7144 _ZN9libcamera13CameraManager7Private3runEv (libcamera.so.0.0.1 + 0x84144)
                                                   #13 0x00007fbd75cd62f3 execute_native_thread_routine (libstdc++.so.6 + 0xd62f3)
                                                   #14 0x00007fbd790428fd n/a (libc.so.6 + 0x868fd)
                                                   #15 0x00007fbd790c3c74 __clone (libc.so.6 + 0x107c74)
                                                   
                                                   Stack trace of thread 54367:
                                                   #0  0x00007fbd7903f4b6 n/a (libc.so.6 + 0x834b6)
                                                   #1  0x00007fbd79041cd0 pthread_cond_wait (libc.so.6 + 0x85cd0)
                                                   #2  0x00007fbd75ccee81 __gthread_cond_wait (libstdc++.so.6 + 0xcee81)
                                                   #3  0x00007fbd75ff35ec _ZN9libcamera13CameraManager7Private5startEv (libcamera.so.0.0.1 + 0x805ec)
                                                   #4  0x00007fbd75ff44fe _ZN9libcamera13CameraManager5startEv (libcamera.so.0.0.1 + 0x814fe)
                                                   #5  0x00007fbd7625abda n/a (libspa-libcamera.so + 0xabda)
                                                   #6  0x00007fbd7625bd0a n/a (libspa-libcamera.so + 0xbd0a)
                                                   #7  0x00007fbd7946a5aa n/a (libwireplumber-0.4.so.0 + 0x265aa)
                                                   #8  0x00007fbd7948e153 wp_transition_advance (libwireplumber-0.4.so.0 + 0x4a153)
                                                   #9  0x00007fbd79473b3a n/a (libwireplumber-0.4.so.0 + 0x2fb3a)
                                                   #10 0x00007fbd792f987b g_main_context_dispatch (libglib-2.0.so.0 + 0x5587b)
                                                   #11 0x00007fbd79350299 n/a (libglib-2.0.so.0 + 0xac299)
                                                   #12 0x00007fbd792f8ddf g_main_loop_run (libglib-2.0.so.0 + 0x54ddf)
                                                   #13 0x000055761672a2fd n/a (wireplumber + 0x22fd)
                                                   #14 0x00007fbd78fdf290 n/a (libc.so.6 + 0x23290)
                                                   #15 0x00007fbd78fdf34a __libc_start_main (libc.so.6 + 0x2334a)
                                                   #16 0x000055761672a5f5 n/a (wireplumber + 0x25f5)
                                                   
                                                   Stack trace of thread 54369:
                                                   #0  0x00007fbd790c4096 epoll_wait (libc.so.6 + 0x108096)
                                                   #1  0x00007fbd794f0669 n/a (libspa-support.so + 0x14669)
                                                   #2  0x00007fbd794e2b7b n/a (libspa-support.so + 0x6b7b)
                                                   #3  0x00007fbd79204317 n/a (libpipewire-0.3.so.0 + 0x41317)
                                                   #4  0x00007fbd790428fd n/a (libc.so.6 + 0x868fd)
                                                   #5  0x00007fbd790c3c74 __clone (libc.so.6 + 0x107c74)
                                                   
                                                   Stack trace of thread 54371:
                                                   #0  0x00007fbd790b70bf __poll (libc.so.6 + 0xfb0bf)
                                                   #1  0x00007fbd793501ff n/a (libglib-2.0.so.0 + 0xac1ff)
                                                   #2  0x00007fbd792f8132 g_main_context_iteration (libglib-2.0.so.0 + 0x54132)
                                                   #3  0x00007fbd792f8182 n/a (libglib-2.0.so.0 + 0x54182)
                                                   #4  0x00007fbd79326e05 n/a (libglib-2.0.so.0 + 0x82e05)
                                                   #5  0x00007fbd790428fd n/a (libc.so.6 + 0x868fd)
                                                   #6  0x00007fbd790c3c74 __clone (libc.so.6 + 0x107c74)
                                                   
                                                   Stack trace of thread 54372:
                                                   #0  0x00007fbd790bc7fd syscall (libc.so.6 + 0x1007fd)
                                                   #1  0x00007fbd79349c17 g_cond_wait_until (libglib-2.0.so.0 + 0xa5c17)
                                                   #2  0x00007fbd792c7893 n/a (libglib-2.0.so.0 + 0x23893)
                                                   #3  0x00007fbd7932c6cb n/a (libglib-2.0.so.0 + 0x886cb)
                                                   #4  0x00007fbd79326e05 n/a (libglib-2.0.so.0 + 0x82e05)
                                                   #5  0x00007fbd790428fd n/a (libc.so.6 + 0x868fd)
                                                   #6  0x00007fbd790c3c74 __clone (libc.so.6 + 0x107c74)
                                                   
                                                   Stack trace of thread 54373:
                                                   #0  0x00007fbd790b70bf __poll (libc.so.6 + 0xfb0bf)
                                                   #1  0x00007fbd793501ff n/a (libglib-2.0.so.0 + 0xac1ff)
                                                   #2  0x00007fbd792f8ddf g_main_loop_run (libglib-2.0.so.0 + 0x54ddf)
                                                   #3  0x00007fbd78eef37c n/a (libgio-2.0.so.0 + 0x10a37c)
                                                   #4  0x00007fbd79326e05 n/a (libglib-2.0.so.0 + 0x82e05)
                                                   #5  0x00007fbd790428fd n/a (libc.so.6 + 0x868fd)
                                                   #6  0x00007fbd790c3c74 __clone (libc.so.6 + 0x107c74)
                                                   ELF object binary architecture: AMD x86-64
```
The entire wireplumber process crashes, preventing any audio from working. This doesn't happen when libcamera isn't installed. This is very detrimental to the user experience. Reproduced elsewhere: https://bbs.archlinux.org/viewtopic.php?id=280902

This has been a known bug for a while so, is probably a dupe but I couldn't find any matching bugs here or on the bugzilla.

kbingham, 2022-12-16:
I believe this is : https://bugs.libcamera.org/show_bug.cgi?id=167

kbingham, 2022-12-16:
This is fixed and merged in commit 96ed45b9711e, and will therefore be included in libcamera v0.0.3.

closed 2022-12-16 by melvyn2