kbingham/libcamera #64: No image format found

kurosh-z, 2023-01-08:
I'm getting the following error on an ov9281 monochrome  sensor when running ./cam --list:

```
./cam --list

[3:16:07.769383485] [7935]  INFO IPAManager ipa_manager.cpp:143 libcamera is not installed. Adding '/home/drone/repos/libcamera/build/src/ipa' to the IPA search path
[3:16:07.779682547] [7935]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4124-5df5b72c
[3:16:07.807202112] [7938] ERROR CameraSensor camera_sensor.cpp:124 'vc_mipi_ov9281 10-0060': No image format found
[3:16:07.808000269] [7938] ERROR RPI raspberrypi.cpp:1228 Failed to register camera vc_mipi_ov9281 10-0060: -22
Available cameras:

```
Any suggestion on how I could fix this?

Camera: https://www.inno-maker.com/product/cam-mipi9281raw-v2/
Driver: https://github.com/INNO-MAKER/CAM-OV9281RAW-V2
Documentation: https://github.com/INNO-MAKER/CAM-OV9281RAW-V2/blob/main/Documents/CAM-MIPIOV9281%20V2%20User%20Manual%20V1.2.pdf

Tested on:
Raspberry pi 4:
OS: "Raspbian GNU/Linux 11 (bullseye)"
Kernel: 5.10.92-v7l+

Regards, 
Kurosh

kbingham, 2023-01-08:
Can you rerun with 'LIBCAMERA_LOG_LEVELS=*:0' set before cam please? Does innomaker have a support channel?

kurosh-z, 2023-01-08:
This is now the cam app with the full debug on:

[image]

I contacted the innomaker and waiting for the answer.

kbingham, 2023-01-08:
I would suspect there is an issue in the innomaker ov9281 driver. Ask them to upstream the driver to the Linux media subsystem of the kernel, and then errors can be identified during the review process.

But isn't there an ov9281 driver in the raspberry pi kernel you can use directly? Does that one work?

kbingham, 2023-01-12:
I have a feeling you might now be using the wrong device tree overlay, as the camera is listed as mov9281. But now you're certainly into Raspberry Pi support territory. Could you raise this issue at github.com/raspberrypi/libcamera please? They can help more. (Or innomaker directly).

kbingham, 2023-01-12:
The upstreamed IPA tuning file in ilbcamera is ./src/ipa/raspberrypi/data/ov9281_mono.json

So I think you can also manually set the tuning file with one of the Raspberry Pi specific environment variables.

closed 2023-07-04 by kbingham