kbingham/libcamera #67: Working on IMX214 on RockPro64- no stream

belveder79, 2023-02-20:
While I was working on the the IMX214 already last year with limited success, I picked it up again and I need some buddies to discuss the problem I'm facing. 

As it seems it is a mixture of device tree problems and problems with the library itself (probably). There is no forum to collect all the findings I made so far, so this is an attempt to do so and make the IMX214 finally work with libcamera - and to provide a working solution that hopefully makes use of libcamera as well.

To start off, here's what I did:

1. the Rockchip-ISP1 driver is compiling nicely on the latest kernels from 5.15.89-rockchip64, so that should provide a working basis. 
2. the libcamera compiles fine, but lacks support for the IMX214. I opened a pull request with the two minor changes (thx to Marvin Schmidt for providing me with this).
3. the mainline IMX214 driver lacks certain capabilities that libcamera requires. Therefore I patched it to fix the missing properties. [Here](https://file.io/If5nHmh3VI8q) is a version with Makefile to compile it outside the kernel branch. Loads fine and finds the camera. I2Ctransfer also does a nice job and gets data out.
5. the device tree itself for the RockPro64 - it's a mess, but I would say it is 95% working...
```
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/pinctrl/rockchip.h>
#include <dt-bindings/clock/rk3399-cru.h>
/ {
	compatible = "rockchip,rk3399";

	fragment@0 {
		target = <&isp0_mmu>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@1 {
		target = <&mipi_dphy_rx0>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@2 {
		target = <&isp0>;
		__overlay__ {
			status = "okay";
			ports {
				port@0 {
					#address-cells = <1>;
					#size-cells = <0>;
					mipi_in_wcam: endpoint@0 {
						reg = <0>;
						remote-endpoint = <&wcam_out>;
						data-lanes = <1 2 3 4>;
					};
				};
			};
		};
	};

	fragment@3 {
		target = <&sdio0>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@4 {
		target = <&i2c1>;
		__overlay__ {

			status = "okay";

			// https://github.com/friendlyarm/kernel-rockchip/blob/nanopi-r2-v5.15.y/arch/arm64/boot/dts/rockchip/rk3399-gru-scarlet.dtsi
			clock-frequency = <400000>;
			i2c-scl-falling-time-ns = <50>;
			i2c-scl-rising-time-ns = <300>;

			// 24M mclk is shared between world and user cameras
			pinctrl-0 = <&i2c1_xfer &cif_clkouta>;
			// pinctrl-1 = <&wcam_rst>;
			#address-cells = <1>;
			#size-cells = <0>;

			wcam: imx214@10 {
				compatible = "sony,imx214";
				reg = <0x10>;

				vdddo-supply = <&vcc1v8_mipi>;
				vddd-supply = <&vcc1v2_mipi_dvdd>;
				vdda-supply = <&vcc2v8_mipi>;

				reset-gpios = <&gpio0 RK_PB4 GPIO_ACTIVE_LOW>; // GPIO0_B4

				clocks = <&cru SCLK_CIF_OUT>;
				clock-names = "xvclk";

				rotation = <180>;
				orientation = <2>;

				// ORIGINAL BEGIN
				port {
					wcam_out: endpoint {
						data-lanes = <1 2 3 4>;
						link-frequencies = /bits/ 64 <480000000>;
						remote-endpoint = <&mipi_in_wcam>;
					};
				};
				// ORIGINAL END
			};
		};
	};

	fragment@5 {
		target = <&pinctrl>;
		__overlay__ {
			dvp {
				dvp_pwr_en: dvp-pwr-en {
					rockchip,pins = <1 RK_PC7 RK_FUNC_GPIO &pcfg_pull_none>;
				};
			};
			camera {
				wcam_rst: wcam_rst {
					rockchip,pins = <0 RK_PB4 RK_FUNC_GPIO &pcfg_pull_none>;
				};
			};
			cif {
				cif_clkouta: cif-clkouta {
						rockchip,pins = <2 RK_PB3 3 &pcfg_pull_up>;
				};
			};
		};
	};

	fragment@6 {
		target-path = "/";
		__overlay__ {
		  vcc2v8_mipi: vcc2v8-mipi-regulator {
		    compatible = "regulator-fixed";
		    enable-active-high;
		    gpio = <&gpio1 RK_PC7 GPIO_ACTIVE_HIGH>;
		    pinctrl-names = "default";
		    pinctrl-0 = <&dvp_pwr_en>;
		    regulator-name = "vcc2v8_mipi";
		    regulator-always-on;
		    regulator-boot-on;
		    regulator-min-microvolt = <2800000>;
		    regulator-max-microvolt = <2800000>;
		    startup-delay-us = <1000>;
		    vin-supply = <&vcc5v0_sys>;
		  };
		  // turned on automatically as vcc2v8_mipi is turned on...
		  vcc1v8_mipi: vcc1v8-mipi-regulator {
		    compatible = "regulator-fixed";
		    regulator-name = "vcc1v8_mipi";
		    regulator-always-on;
		    regulator-boot-on;
		    regulator-min-microvolt = <1800000>;
		    regulator-max-microvolt = <1800000>;
		    startup-delay-us = <1000>;
		    vin-supply = <&vcc3v3_sys>;
		  };

		  vcc2v8_mipi_af: vcc2v8-mipi-af-regulator {
		    compatible = "regulator-fixed";
		    regulator-name = "vcc2v8_mipi_af";
		    regulator-always-on;
		    regulator-boot-on;
		    regulator-min-microvolt = <2800000>;
		    regulator-max-microvolt = <2800000>;
		    startup-delay-us = <1000>;
		    vin-supply = <&vcc5v0_sys>;
		  };

		  vcc1v2_mipi_dvdd: vcc1v2-mipi-dvdd-regulator {
		    compatible = "regulator-fixed";
		    regulator-name = "vcc1v2_mipi_dvdd";
		    regulator-always-on;
		    regulator-boot-on;
		    regulator-min-microvolt = <1200000>;
		    regulator-max-microvolt = <1200000>;
		    startup-delay-us = <1000>;
		    vin-supply = <&vcc3v3_sys>;
		  };
		};
	};
};
```

I originally had some trouble setting up the clocks, but those seem to be fine thx to a pro from the Discord forum. According to [the specs](https://files.pine64.org/doc/rockpro64/rockpro64_v21-SCH.pdf) there is GPIO2_B4 and GPIO0_B4 for the PowerDown-Pins, depending on the MIPI port used, but I can't really make sense out of them.

Anyway, here's the issue: the libcamera works and finds the camera, but stream mapping is broken (it gets stuck). Similarly lc_compliance gets stuck already at the first test. There is also this media-ctl sequence, which should get some data out at some point:

```
"media-ctl" "-d" "platform:rkisp1" "-l" "'imx214 1-0010':0 -> 'rkisp1_isp':0 [1]"
"media-ctl" "-d" "platform:rkisp1" "-l" "'rkisp1_isp':2 -> 'rkisp1_resizer_selfpath':0 [1]"
"media-ctl" "-d" "platform:rkisp1" "-l" "'rkisp1_isp':2 -> 'rkisp1_resizer_mainpath':0 [0]"
"media-ctl" "-d" "platform:rkisp1" "--set-v4l2" '"imx214 1-0010":0 [fmt:SRGGB10_1X10/1920x1080]'
"media-ctl" "-d" "platform:rkisp1" "--set-v4l2" '"rkisp1_isp":0 [fmt:SRGGB10_1X10/1920x1080 crop: (0,0)/1920x1080]'
"media-ctl" "-d" "platform:rkisp1" "--set-v4l2" '"rkisp1_isp":2 [fmt:YUYV8_2X8/1920x1080 crop: (0,0)/1920x1080]'
"media-ctl" "-d" "platform:rkisp1" "--set-v4l2" '"rkisp1_resizer_selfpath":0 [fmt:YUYV8_2X8/1920x1080 crop: (0,0)/1920x1080]'
"media-ctl" "-d" "platform:rkisp1" "--set-v4l2" '"rkisp1_resizer_selfpath":1 [fmt:YUYV8_2X8/1920x1080]'
"media-ctl" "-d" "platform:rkisp1" "--set-v4l2" '"rkisp1_resizer_mainpath":0 [fmt:YUYV8_2X8/1920x1080 crop: (0,0)/1920x1080]'
"media-ctl" "-d" "platform:rkisp1" "--set-v4l2" '"rkisp1_resizer_mainpath":1 [fmt:YUYV8_2X8/1920x1080]'
"v4l2-ctl" "-z" "platform:rkisp1" "-d" "rkisp1_selfpath" "-v" "width=1920,height=1080,"
"v4l2-ctl" "-z" "platform:rkisp1" "-d" "rkisp1_selfpath" "-v" "pixelformat=422P"
```

and ultimately

```
"v4l2-ctl" "-z" "platform:rkisp1" "-d" "rkisp1_selfpath" "--stream-mmap" "--stream-count" "1" --stream-to=single.raw
```

fails with "broken pipe" or "invalid device" or something similar. So it seems that libcamera has the same problem with the raw stream capture by media-ctl/video-ctl..., and I can't get rid of the feeling that I'm missing something somewhere. There was a conversation on the DietPi forum about [this](https://github.com/MichaIng/DietPi/issues/3360), but no real working solution at the end either... 

I have a similar setup for the OV13850 as well, for which I also already patched the devicetree and libcamera - that's a slightly different story, but the outcome is the same... will try to summarize this in an other submission...

kbingham, 2023-02-20:
please report the logs from:

`LIBCAMERA_LOG_LEVELS=*:0 cam -c1 -C100`

(Capture 100 frames from the first camera)

belveder79, 2023-02-20:
The output is

```
# LD_LIBRARY_PATH=~/libcamera-install/lib/aarch64-linux-gnu LIBCAMERA_LOG_LEVELS=*:0 /root/libcamera-install/bin/cam -c1 -C100

[0:02:40.039595825] [1248] DEBUG IPAModule ipa_module.cpp:329 ipa_rkisp1.so: IPA module /root/libcamera-install/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so is signed
[0:02:40.040219992] [1248] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/root/libcamera-install/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so'
[0:02:40.042335742] [1248] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /root/libcamera-install/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so is signed
[0:02:40.042791617] [1248] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/root/libcamera-install/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so'
[0:02:40.043682075] [1248]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+16-2403f514-dirty (2023-02-20T17:22:27+00:00)
[0:02:40.044475742] [1249] DEBUG Camera camera_manager.cpp:108 Starting camera manager
[0:02:40.060605783] [1249] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "hantro-vpu" created from /dev/media1
[0:02:40.061164617] [1249] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media1 due to 2 missing dependencies
[0:02:40.063264325] [1249] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media1 found
[0:02:40.063355908] [1249] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: hantro-vpu
[0:02:40.064557575] [1249] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "rkvdec" created from /dev/media0
[0:02:40.064720033] [1249] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media0 due to 1 missing dependencies
[0:02:40.065613992] [1249] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media0 found
[0:02:40.065676700] [1249] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: rkvdec
[0:02:40.067020992] [1249] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "rkisp1" created from /dev/media2
[0:02:40.067716325] [1249] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media2 due to 7 missing dependencies
[0:02:40.072302200] [1249] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media2 found
[0:02:40.072410700] [1249] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: rkisp1
[0:02:40.073111283] [1249] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerISI'
[0:02:40.073884200] [1249] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerRPi'
[0:02:40.074085450] [1249] DEBUG RPI raspberrypi.cpp:1254 Unable to acquire a Unicam instance
[0:02:40.074147867] [1249] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerRkISP1'
[0:02:40.074700283] [1249] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "rkisp1"
[0:02:40.075686117] [1249] DEBUG V4L2 v4l2_videodevice.cpp:632 /dev/video7[14:cap]: Opened device platform:rkisp1: rkisp1: rkisp1_stats
[0:02:40.076908742] [1249] DEBUG V4L2 v4l2_videodevice.cpp:632 /dev/video8[15:out]: Opened device platform:rkisp1: rkisp1: rkisp1_params
[0:02:40.077648117] [1249] DEBUG V4L2 v4l2_videodevice.cpp:632 /dev/video5[17:cap]: Opened device platform:rkisp1: rkisp1: rkisp1
[0:02:40.078582908] [1249] ERROR V4L2 v4l2_videodevice.cpp:1183 /dev/video5[17:cap]: Unable to enumerate frame sizes: Inappropriate ioctl for device
[0:02:40.078900825] [1249]  INFO RkISP1 rkisp1_path.cpp:91 Failed to enumerate supported formats and sizes, using defaults
[0:02:40.079752492] [1249] DEBUG V4L2 v4l2_videodevice.cpp:632 /dev/video6[19:cap]: Opened device platform:rkisp1: rkisp1: rkisp1
[0:02:40.080396200] [1249] ERROR V4L2 v4l2_videodevice.cpp:1183 /dev/video6[19:cap]: Unable to enumerate frame sizes: Inappropriate ioctl for device
[0:02:40.080557492] [1249]  INFO RkISP1 rkisp1_path.cpp:91 Failed to enumerate supported formats and sizes, using defaults
[0:02:40.083204950] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Exposure (0x00980911)
[0:02:40.084529992] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Horizontal Flip (0x00980914)
[0:02:40.084931617] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Vertical Flip (0x00980915)
[0:02:40.085188867] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Camera Orientation (0x009a0922)
[0:02:40.085703367] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Camera Sensor Rotation (0x009a0923)
[0:02:40.085959450] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Vertical Blanking (0x009e0901)
[0:02:40.086180825] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Horizontal Blanking (0x009e0902)
[0:02:40.086414158] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Analogue Gain (0x009e0903)
[0:02:40.086636783] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Red Pixel Value (0x009e0904)
[0:02:40.086763950] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Green (Red) Pixel Value (0x009e0905)
[0:02:40.086878575] [1249] DEBUG V4L2 v4l2_device.cpp:635 'imx214 1-0010': Control: Blue Pixel Value (0x009e0906)
[... cut ...]
[0:02:42.333010284] [1249] ERROR V4L2 v4l2_videodevice.cpp:1697 /dev/video5[17:cap]: Failed to queue buffer 3: Invalid argument
[0:02:42.333422701] [1249] DEBUG V4L2 v4l2_videodevice.cpp:1764 /dev/video8[15:out]: Dequeuing buffer 0
[0:02:42.334106659] [1249] DEBUG V4L2 v4l2_videodevice.cpp:1764 /dev/video8[15:out]: Dequeuing buffer 1
```

and it comes to a halt (and the camera makes a strange click noise once)...

what I could also see is this on the kernel log

``` 
[  269.371078] imx214 1-0010: ctrl(id:0x980914,val:0x1) is not handled
[  269.371094] imx214 1-0010: could not sync v4l2 controls
```

kbingham, 2023-02-20:
Seems like a fault setting the controls on the subdevice gets things upset? I'm not sure why that causes buffers to stop queueing though. Is there a patch that shows any changes you've made to the imx214 ?