kbingham/libcamera #71: question

Arnold1, 2023-03-17:
hi,

does it also work on raspberry pi zero 2 w?

kbingham, 2023-03-17:
Yes, libcamera is functional and supported on the RPi zero 2 W with all of the Raspberry Pi Cameras.

Indeed, the Camera v3 is the IMX708. What is your specific use case for putting it to sleep? You can stop the camera, and restart it again later. It will be up to the drivers to ensure that it is powered down when stopped.

If you have issues with this, you can either report this to Raspberry Pi - or contact us if you require specific kernel investigations and development.

--
Kieran

Arnold1, 2023-03-18:
I want to build some surveillance system. If i could sleep the camera - i would safe lots of energy…

kbingham, 2023-03-18:
You can call camera::start() and camera::stop(). But it's possible that it currently won't likely save much power.

That's something we could work on to improve if you need it. You could already measure power consumptions on your device to see if this if your configuration has runtime power management enabled.

Arnold1, 2023-03-18:
Hi @kbingham would love to see a power saving feature when the camera is stopped - how would that work on the driver / lib?

kbingham, 2023-03-20:
The OV5640 driver already supports runtime-pm. https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ov5640.c#n3705

That means calling camera::stop() should enable the power saving mode. If you are finding that the power consumption is still higher than you expect, then we could look at setting up a project to investigate this for you.

Arnold1, 2023-03-21:
yes i know - but not for the raspberry pi camera module 3? there is no good way to save power consumption using raspberry pi camera module 3?

kbingham, 2023-03-22:
The IMX708 driver also has runtime-pm. Can you clarify exactly what you have measured to determine there's a fault here?

 - https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/media/i2c/imx708.c#L1489

closed 2023-07-04 by kbingham