kbingham/libcamera #72: not able to build from source

toor11, 2023-04-08:
user@pop-os:~/Downloads/libcamera/libcamera$ sudo ninja -C build install
ninja: Entering directory `build'
[1/2] Installing files.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mesonbuild/mesonmain.py", line 146, in run
    return options.run_func(options)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 811, in run
    installer.do_install(datafilename)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 564, in do_install
    d = self.check_installdata(pickle.load(ifile))
AttributeError: Can't get attribute 'OptionString' on <module 'mesonbuild.interpreter.primitives.string' from '/usr/lib/python3/dist-packages/mesonbuild/interpreter/primitives/string.py'>

ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
FAILED: meson-internal__install 
/usr/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.

kbingham, 2023-04-09:
Sounds like meson is suggesting a meson bug, which should be reported to https://github.com/mesonbuild/meson but I expect checking you've got a recent version of meson first would be worthwhile.

What does the full configuration log state?

kbingham, 2023-04-09:
Oh also I think this is often a case of a different meson version when used with sudo Vs non-sudo.

Check with 

'meson --version'
And compare against
'sudo meson --version'

eli-schwartz, 2023-04-09:
> Oh also I think this is often a case of a different meson version when used with sudo Vs non-sudo.

That's correct.

> Sounds like meson is suggesting a meson bug, which should be reported to https://github.com/mesonbuild/meson but I expect checking you've got a recent version of meson first would be worthwhile.

In this case, the bug is fixed by newer versions of Meson. Unfortunately, even installing newer versions of Meson don't help if you run sudo, which then uses a really old version instead.

closed 2023-05-24 by kbingham