kbingham/libcamera #74: Installing libcamera from source on docker container succeeds but cannot import libcamera into python

habjoel, 2023-05-03:
Hey there,
I am trying to get libcamera to work inside my docker container that is based on the `arm64v8/python3.9-bullseye` docker image. My ultimate goal is to get libcamera working inside a docker container on my Raspberry Pi CM4.

I have gone through the installation steps specified [here](https://libcamera.org/getting-started.html). and I used the following options with meson:
```sh
meson build --buildtype=release -Dpipelines=raspberrypi -Dipas=raspberrypi -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dcam=enabled -Dqcam=disabled -Dpycamera=enabled
```
The installation did not throw any errors and I built the python bindings by enabling pycamera. However I cannot really import libcamera into my python. `import libcamera` somehow only works if I am under root directory (`/`) and I cannot import any submodules such as `from libcamera import controls` throws an `ImportError: cannot import name 'controls' from 'libcamera' (unknown location)`. From every other location nothing works (`ModuleNotFoundError: No module named 'libcamera'`). Moreover, I can only find the header files of libcamera under `/usr/local/include` but no source files are under `/usr/local/src`.

Does anyone have a clue how to solve this? Obviously, I installed everything as a root user as this is the default inside Docker containers. Is that a problem?  Moreover, I am using a python 3.9 specific docker image so this should be the only python inside the container.

Thank you for any help!

kbingham, 2023-05-03:
I'm not sure what the python search paths are. Google to find out how to print those on your system, then match up where you're installing the libcamera package.
On a build here for instance it's gone to :

 - /usr/lib/x86_64-linux-gnu/python3.10/site-packages/libcamera

You might want to add -Dprefix=/usr on your build to install to /usr or add /usr/local/lib/..... to your python search paths.

habjoel, 2023-05-08:
thanks a lot. It worked by adding `-Dprefix=/usr` and adding `/usr/lib/aarch64-linux-gnu/python3.9/site-packages/` to my python path.
In addition, I need to use picamera2. I installed it onto my docker container using pip install picamera2. The resulting version is 0.3.9. As I import it into python (`import picamera2`), it throws the following error:
```
ImportError: cannot import name 'convert_from_libcamera_type' from 'picamera2.utils' (/usr/local/lib/python3.9/site-packages/picamera2/utils/__init__.py)
``` 
Might there be some kind of version mismatch between picamera2 and libcamera? 

Thanks for any help!

kbingham, 2023-05-12:
I was going to reply to say if you're having issues with picamera2 - it's better to ask on that project. It's separate, but it looks like you already have!

Sounds like you've worked out the issue over there so closing this.

closed 2023-05-12 by kbingham