kbingham/libcamera #76: Failed to get Encoded video data from camera after integrating libcamera SPA in pipewire

MukeshkumarK, 2023-05-24:
Hi,
We are integrating pipewire with libcamera SPA plugin enabled use case in Raspberrypi 3B+ target with RPI CSI camera.
Our main intention of this integration is multiple client application need's to get h264 encoded data for streaming and recording with libcamera as camera background.
Receiving camera buffers properly while streaming in VLC player with v4l2 SPA but , Not able to get buffer when using libcamera SPA enabled.

MukeshkumarK, 2023-05-24:
`--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 24-05-2023 (09:08:03)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi3-rdk-camera

** cat /proc/cpuinfo **
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000086d1c8b2
Model		: Raspberry Pi 3 Model B Plus Rev 1.3

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait    

** cat /boot/config.txt **
################################################################################
##  Raspberry Pi Configuration Settings
##
##  Revision 16, 2013/06/22
##
[... 1898 lines cut ...]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

--------------------------------------------------------------------------------
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi3-rdk-camera 5.15.76-v7 #1 SMP Wed May 24 07:32:14 UTC 2023 armv7l GNU/Linux

** cat /etc/os-release **
ID=rdk
NAME=RDK (A Yocto Project based Distro)
VERSION=2.0 (dunfell)
VERSION_ID=2.0
PRETTY_NAME=RDK (A Yocto Project based Distro) 2.0 (dunfell)

** vcgencmd version **
/bin/sh: vcgencmd: command not found

** libcamera-hello --version **
libcamera-apps build: 
libcamera build: v0.0.1
`

kbingham, 2023-11-06:
closing as libcamera is not an encoder library. libcamera deals with image processing of the camera.

closed 2023-11-06 by kbingham