kbingham/libcamera #77: OV5647 Unable to find ISP control 0x009819e1

614674490, 2023-06-25:
In my system, when I downgraded `libcamera` to version 0.0.0 and ran `libcamera-hello`, it returned the error message `failed to configure streams.`
version: libcamera: 0.0.0(commit: 5acc21f), libacmera-apps: 1.0.0(commit:commit: f4070274)
```sh
rtos@raspberrypi:~ $ ls /dev/video*
/dev/video0  /dev/video1  /dev/video10  /dev/video11  /dev/video12  /dev/video13  /dev/video14  /dev/video15  /dev/video16  /dev/video18
rtos@raspberrypi:~ $  vcgencmd get_camera
supported=1 detected=0, libcamera interfaces=1
rtos@raspberrypi:~ $  libcamera-hello
Preview window unavailable
[0:25:13.852641086] [13826]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:25:13.966089574] [13826]  INFO Camera camera.cpp:945 configuring streams: (0) 1296x972-YUV420
[0:25:13.966784505] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 2067 (best 2067)
[0:25:13.966884043] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1849.67 (best 1849.67)
[0:25:13.966924302] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 1500 (best 1500)
[0:25:13.966960858] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 3796 (best 1500)
[0:25:13.967001840] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 1567 (best 1500)
[0:25:13.967038377] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1349.67 (best 1349.67)
[0:25:13.967074044] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 1000 (best 1000)
[0:25:13.967109229] [13828]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 3296 (best 1000)
[0:25:13.967370509] [13828]  INFO RPI raspberrypi.cpp:632 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1296x972-pBAA
[0:25:13.969421172] [13828] ERROR IPARPI raspberrypi.cpp:567 Unable to find ISP control 0x009819e1
[0:25:13.969504783] [13828] ERROR IPARPI raspberrypi.cpp:356 ISP control validation failed.
[0:25:13.969532469] [13828] ERROR RPI raspberrypi.cpp:1298 IPA configuration failed!
[0:25:13.969633488] [13828] ERROR RPI raspberrypi.cpp:765 Failed to configure the IPA: -32
ERROR: *** failed to configure streams ***
```
I have enabled the camera in `raspi-config`, as shown in the screenshot below.
[image]

/boot/config.txt内容如下
```txt
start_x = 1
[all]
gpu_mem = 128
dtoverlay=ov5647
```
My camera is a first-generation OV5647 model, and I'm not sure why this error is occurring.

kbingham, 2023-06-26:
Why are you trying to downgrade your libcamera installation? Is there a reason you can't use the latest libcamera ?

614674490, 2023-06-26:
We are currently using the 5.13 version of the kernel system (`Linux raspberrypi 5.13.19-v8+ #7 SMP PREEMPT IRQPIPE Wed May 24 16:59:45 CST 2023 aarch64 GNU/Linux`). The camera driver in this system requires the use of V4L2 and does not support the media controller. Additionally, higher versions of libcamera rely on a higher version of unicam, but the version of unicam we are using in our system is lower. Therefore, we need to downgrade the libcamera version to 0.0.0. However, due to the large number of commits in your project, we have not found a suitable version yet.
We have already ported our own RTOS system on this system. When using the libcamera provided by the 5.13 kernel, we are unable to access the camera. We have identified that the issue is related to the higher versions of libcamera using the media controller to control the camera. Hence, we need to downgrade to a specific commit of version 0.0.0.

614674490, 2023-06-26:
We require a version of `libcamera` that does not rely on the `media controller`  to install in our 5.13 kernel system. If we were to introduce the` media controller` into our system to support higher versions of libcamera, it would involve incorporating over 100 commits from the Raspberry Pi community and the Linux community. This would create further complications. Therefore, we need to downgrade `libcamera` to a suitable version in order to use the camera.

kbingham, 2023-06-26:
I'm really concerned that this sounds like the wrong approach. libcamera is incredibly tied to using media-controller devices to identify what camera to use. 

libcamera v0.0.0 - is really old, and represents a much earlier development point, which is sort of what we try to express by the number '0' there.

You also mention you're running your own RTOS ... are you trying to use libcamera in a non-linux OS ?

kbingham, 2023-06-26:
To be clear, I would recommend that you go the other way - use the latest kernel from Raspberry Pi - and the latest libcamera.

kbingham, 2023-06-26:
It was possible to run on the early kernels while the development was being added - but it is not feature complete, and crucially - those earlier versions *are not supported*. There is no continuation of development on any prior version of libcamera. Only the latest version is supportable and under development.

If you have a commercial requirement to get libcamera running on the older kernels, please contact me via e-mail and we can help support you with the backporting of kernel drivers where necessary to be able to run the latest libcamera.

Richardhongyu, 2023-06-26:
Thanks for your help! We will try to backport the libcamera and contact you for your help when necessary.

614674490, 2023-07-01:
Running `v4l2-ctl -d 13 --list-ctrls # bcm2835-isp` on our system, the output is as follows
```log
User Controls

                    red_balance 0x0098090e (int)    : min=1 max=65535 step=1 default=1000 value=1482 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=65535 step=1 default=1000 value=1378 flags=slider
       colour_correction_matrix 0x009819f1 (u8)     : min=0 max=255 step=1 default=0 [88] flags=has-payload
                   lens_shading 0x009819f2 (u8)     : min=0 max=255 step=1 default=0 [36] flags=has-payload, execute-on-write
                    black_level 0x009819f3 (u8)     : min=0 max=255 step=1 default=0 [12] flags=has-payload
             green_equalisation 0x009819f4 (u8)     : min=0 max=255 step=1 default=0 [16] flags=has-payload
                          gamma 0x009819f5 (u8)     : min=0 max=255 step=1 default=0 [136] flags=has-payload
                        denoise 0x009819f6 (u8)     : min=0 max=255 step=1 default=0 [24] flags=has-payload
                        sharpen 0x009819f7 (u8)     : min=0 max=255 step=1 default=0 [28] flags=has-payload
     defective_pixel_correction 0x009819f8 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload
                 colour_denoise 0x009819f9 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload

Image Processing Controls

                   digital_gain 0x009f0905 (int)    : min=1 max=65535 step=1 default=1000 value=1000
```
but the output in the `2021-10-30-raspios-bullseye-arm64-lite` system is
```log
User Controls

                    red_balance 0x0098090e (int)    : min=1 max=65535 step=1 default=1000 value=1000 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=65535 step=1 default=1000 value=1000 flags=slider
       colour_correction_matrix 0x009819e1 (u8)     : min=0 max=255 step=1 default=0 [88] flags=has-payload
                   lens_shading 0x009819e2 (u8)     : min=0 max=255 step=1 default=0 [36] flags=has-payload, execute-on-write
                    black_level 0x009819e3 (u8)     : min=0 max=255 step=1 default=0 [12] flags=has-payload
             green_equalisation 0x009819e4 (u8)     : min=0 max=255 step=1 default=0 [16] flags=has-payload
                          gamma 0x009819e5 (u8)     : min=0 max=255 step=1 default=0 [136] flags=has-payload
                        denoise 0x009819e6 (u8)     : min=0 max=255 step=1 default=0 [24] flags=has-payload
                        sharpen 0x009819e7 (u8)     : min=0 max=255 step=1 default=0 [28] flags=has-payload
     defective_pixel_correction 0x009819e8 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload
                 colour_denoise 0x009819e9 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload

Image Processing Controls

                   digital_gain 0x009f0905 (int)    : min=1 max=65535 step=1 default=1000 value=1000
```
The reason why `Unable to find ISP control 0x009819e1` is because the address of our system is 0x009819f1, so the address of [v4l2_controls.h](https://github.com/raspberrypi/libcamera/blob/ 5acc21fd04fa8568eea9a0bc59e0e614acfbaf7c/include/linux/v4l2-controls.h#L215) in line 214 of code `#define V4L2_CID_USER_BCM2835_ISP_BASE (V4L2_ CID_USER_BASE + 0x10e0)` to `#define V4L2_CID_USER_BCM2835_ISP_BASE (V4L2_CID_USER_BASE + 0x10f0)` **
Then run `libcamera-hello` and the output is as follows
```log
[0:34:43.162964315] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Saturation = 1.000000
[0:34:43.163062853] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Contrast = 1.000000
[0:34:43.163094007] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Brightness = 0.000000
[0:34:43.163123161] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: AwbMode = 0
[0:34:43.163151686] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Sharpness = 1.000000
[0:34:43.163204418] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: ExposureValue = 0.000000
[0:34:43.163260504] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: AeExposureMode = 0
[0:34:43.163289343] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:34:43.163324331] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: AeMeteringMode = 0
[0:34:43.163351541] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: NoiseReductionMode = 3
[0:34:43.164066888] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.164153442] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.164202674] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.164249369] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.266818054] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[0:34:43.311877230] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:34:43.345672145] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:34:43.379652504] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:34:43.411950044] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:34:43.448440074] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:34:43.478996865] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:34:43.513571205] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
```
This still seems to have some problems, but currently captures pictures and videos properly

614674490, 2023-07-01:
Running `v4l2-ctl -d 13 --list-ctrls # bcm2835-isp` on our system, the output is as follows
```log
User Controls

                    red_balance 0x0098090e (int)    : min=1 max=65535 step=1 default=1000 value=1482 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=65535 step=1 default=1000 value=1378 flags=slider
       colour_correction_matrix 0x009819f1 (u8)     : min=0 max=255 step=1 default=0 [88] flags=has-payload
                   lens_shading 0x009819f2 (u8)     : min=0 max=255 step=1 default=0 [36] flags=has-payload, execute-on-write
                    black_level 0x009819f3 (u8)     : min=0 max=255 step=1 default=0 [12] flags=has-payload
             green_equalisation 0x009819f4 (u8)     : min=0 max=255 step=1 default=0 [16] flags=has-payload
                          gamma 0x009819f5 (u8)     : min=0 max=255 step=1 default=0 [136] flags=has-payload
                        denoise 0x009819f6 (u8)     : min=0 max=255 step=1 default=0 [24] flags=has-payload
                        sharpen 0x009819f7 (u8)     : min=0 max=255 step=1 default=0 [28] flags=has-payload
     defective_pixel_correction 0x009819f8 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload
                 colour_denoise 0x009819f9 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload

Image Processing Controls

                   digital_gain 0x009f0905 (int)    : min=1 max=65535 step=1 default=1000 value=1000
```
But the output on the `2021-10-30-raspios-bullseye-arm64-lite` system is
```log
User Controls

                    red_balance 0x0098090e (int)    : min=1 max=65535 step=1 default=1000 value=1000 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=65535 step=1 default=1000 value=1000 flags=slider
       colour_correction_matrix 0x009819e1 (u8)     : min=0 max=255 step=1 default=0 [88] flags=has-payload
                   lens_shading 0x009819e2 (u8)     : min=0 max=255 step=1 default=0 [36] flags=has-payload, execute-on-write
                    black_level 0x009819e3 (u8)     : min=0 max=255 step=1 default=0 [12] flags=has-payload
             green_equalisation 0x009819e4 (u8)     : min=0 max=255 step=1 default=0 [16] flags=has-payload
                          gamma 0x009819e5 (u8)     : min=0 max=255 step=1 default=0 [136] flags=has-payload
                        denoise 0x009819e6 (u8)     : min=0 max=255 step=1 default=0 [24] flags=has-payload
                        sharpen 0x009819e7 (u8)     : min=0 max=255 step=1 default=0 [28] flags=has-payload
     defective_pixel_correction 0x009819e8 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload
                 colour_denoise 0x009819e9 (u8)     : min=0 max=255 step=1 default=0 [8] flags=has-payload

Image Processing Controls

                   digital_gain 0x009f0905 (int)    : min=1 max=65535 step=1 default=1000 value=1000
```
The reason why `Unable to find ISP control 0x009819e1` is because it is `0x009819f1` on our system. So change `0x10e0` to `0x10f0` in line 215 of v4l2-controls.h((https://github.com/raspberrypi/libcamera/blob/ 5acc21fd04fa8568eea9a0bc59e0e614acfbaf7c/include/linux/v4l2-controls.h#L215), and then run `libcamera-hello` and the output is as followsso
```log
[0:34:43.162964315] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Saturation = 1.000000
[0:34:43.163062853] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Contrast = 1.000000
[0:34:43.163094007] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Brightness = 0.000000
[0:34:43.163123161] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: AwbMode = 0
[0:34:43.163151686] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: Sharpness = 1.000000
[0:34:43.163204418] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: ExposureValue = 0.000000
[0:34:43.163260504] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: AeExposureMode = 0
[0:34:43.163289343] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:34:43.163324331] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: AeMeteringMode = 0
[0:34:43.163351541] [27917]  INFO IPARPI raspberrypi.cpp:633 Request ctrl: NoiseReductionMode = 3
[0:34:43.164066888] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.164153442] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.164202674] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.164249369] [27913]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:34:43.266818054] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[0:34:43.311877230] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:34:43.345672145] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:34:43.379652504] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:34:43.411950044] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:34:43.448440074] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:34:43.478996865] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:34:43.513571205] [27913]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
```
This seems to have some problems, but captures pictures and videos properly

kbingham, 2023-07-04:
Ah yes - because the controls weren't upstreamed, the linux kernel added a new control in the meanwhile so the underlying values had to change.

I'm happy to hear you've resolved the issue.

closed 2023-07-04 by kbingham