kbingham/libcamera #78: cannot stream an Arducam Hawkeye using libcamera

LordDarkneo, 2023-07-07:
Hello there,

I am tryong to use my Haweye RPI Cam with live streaming functionnality, but I am facing an error few seconds after the start of the streaming flow.

The error shown on the terminal is 

> [0:33:55.851736159] [1601]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/arducam_64mp@1a - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected unicam format: 1920x1080-pRAA
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  failed to import fd 22
> Abandon

The way I am using the stream flow is quite easy:
1- run the SSH command libcamera-vid -t 0 --width 1920 --height 1080 --codec h264 --inline --listen -o tcp://0.0.0.0:8888
2- On the client, in VLC, open the network flow tcp/h264://raspberrypi.local:8888

The video is shown like on second, and the the command crashes on the RPI.

Thanks for your help!

kbingham, 2023-07-10:
I'm sorry I can't provide a better answer here - but the Arducam Hawkeye is not upstreamed by Arducam and therefore not supported by us in the libcamera project.

The only advice I can offer for now is to please raise your issue with Arducam directly.

LordDarkneo, 2023-07-10:
Hi and thanks for your reply. 
That's quite strange because as of now, I am using the RaspController app (on Android) and they have 3 way of streaming video (Legacy, Libcamera & USB). And when I use the Libcamera one, it is perfectly working... The major pain point is that I must be on my local network to use it (I need to check if port forwarding is compatible with the app).
cheers

kbingham, 2023-07-10:
If you can find the same issue on a Raspberry Pi official camera - you can get support direct from Raspberry Pi @ https://github.com/raspberrypi/libcamera/issues/ or at their camera forum https://forums.raspberrypi.com/viewforum.php?f=43