kbingham/libcamera #79: Unable to locate package https://git.linuxtv.org/libcamera

onelowzx6r, 2023-08-18:
Linux Mint surface pro 6 trying to follow the directions but gets stuck here

surface@Surface-Pro-6:~$ sudo apt install git https://git.linuxtv.org/libcamera/libcamera.git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package https://git.linuxtv.org/libcamera

onelowzx6r, 2023-08-18:
sorry this was my initial try

surface@Surface-Pro-6:~$ sudo apt install git https://git.libcamera.org/libcamera/libcamera.git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package https://git.libcamera.org/libcamera

kbingham, 2023-08-19:
Those aren't package locations they are git repositories.

'sudo apt install libcamera'

Is probably the right command but what are you actually trying to achieve?

closed 2023-08-28 by kbingham