kbingham/libcamera #82: Contrast enhancement on/off using controls

Pepslee, 2023-09-09:
Is there a way to turn on/off  automatic contrast enhancement on the fly, but not just in config file ?
It is really great feature, but now I have to modify imx477.json file and change ce_enable value every time.

kbingham, 2023-09-10:
If there isn't already it could be proposed to the development list.

It would need the control to be defined, and then plumbing it would likely be quite easy if there's already a switch in the tuning files anyway

Pepslee, 2023-09-10:
> If there isn't already it could be proposed to the development list.
> 
> It would need the control to be defined, and then plumbing it would likely be quite easy if there's already a switch in the tuning files anyway

How could I do this ? To propose to add this feature to development list ?

kbingham, 2023-09-11:
https://libcamera.org/contributing.html

kbingham, 2023-09-11:
Now logged at : https://bugs.libcamera.org/show_bug.cgi?id=198

closed 2023-09-11 by kbingham