kbingham/libcamera #83: V4L2 Issues?? Noir v3 wide

MattGeale, 2023-09-23:
Hi all,
I'm stumped and have no idea where to go from here.

I have a PI 4B 8gb, Noir v3 wide camera and a fresh install of Full Raspian 64bit. Whenever I try and run libcamera-still -o test.jpg or something along those lines, libcamera complains and throws out errors such as:
```
[0:18:18.676643661] [1827] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
```
[multiple errors in sequence vs just a snippet](https://pastebin.com/gmvc1rXX)

I've done full upgrade, apt update, and also verified the camera is being acknowledged as when running the libcamera-hello --list-cameras I get: 
```
pi@raspberrypi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx708_wide_noir [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

pi@raspberrypi:~ $
``` 
Libcamera-hello -t 0 when running the command via on the pi terminal (instead of ssh), shows the camera window briefly too per design of libcamera-hello, but those errors are still present.

If anyone could let me know what im doing wrong or if I'm missing some crucial part of configuring libcamera to play nice.

Thanks in advance!

kbingham, 2023-09-24:
Closing this as it should be discussed with raspberry pi, and I see you've already reported at their repo.

closed 2023-09-24 by kbingham

MattGeale, 2023-09-24:
> Closing this as it should be discussed with raspberry pi, and I see you've already reported at their repo.
> 

Copy, thanks. Yeah I raised with them because I assumed both repos were the same as they're exact clones of each other. Will update this thread too when and if I solve. Thanks!