kbingham/libcamera #86: A reasonably simple way to bundle libcamera, e. g. inside AppImage?

VioletGiraffe, 2023-12-10:
Hi! I have an application made with libcamera and I want to distribute it in a way that would work across slightly different OS versions and configs. I usually use AppImage for such tasks, it's a great tool and works very well. But libcamera is giving me a lot of grief. Let me explain.

The way it works is I run `AppImageTool` on my executable, it transitively / recursively calls `ldd` on the executable and each reported library, and deploys them all in a folder. The original paths are preserved, but now it's `./AppDir/usr/local/lib/aarch64-linux-gnu/libcamera/` instead of `/usr/local/lib/aarch64-linux-gnu/libcamera`, and so on. At runtime this folder structure is mounted inside `/tmp/`, and it either patches something, and/or uses a modified loader to actually load all the dependencies from this tmp path.

Bundling libcamera.so is no problem, `ldd` shows it as a dependency of my executable, so it gets bundled, along with all of its direct dependencies (direct = shown by `ldd`). The problem is with IPAs. When I run this AppImage on a clean system, I get IPA load error and nothing works. For the obvious reason that IPAs are loaded at runtime, so not reported by `ldd`, and thus not bundled. Okay, no problem. I study the source code and find the `LIBCAMERA_IPA_MODULE_PATH` environment variable which I can use to tell libcamera to find IPAs inside the correct `/tmp/...` directory, and I deploy these libraries manually. But then doesn't work with the following errors:

```
[4:26:37.694544779] [4807] ERROR IPCPipe ipc_pipe_unixsocket.cpp:134 Call timeout!
[4:26:37.695036974] [4807] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[4:26:37.695124163] [4807] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:198 Failed to call init
[4:26:37.695250520] [4807] ERROR RPI pipeline_base.cpp:802 Failed to load a suitable IPA library
[4:26:37.695328646] [4807] ERROR RPI vc4.cpp:216 Failed to register camera imx477 10-001a: -22
[4:26:37.695932302] [4807] ERROR IPCUnixSocket ipc_unixsocket.cpp:191 Failed to send: Connection refused
[4:26:37.696273662] [4807] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[4:26:37.696340329] [4807] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:470 Failed to call unmapBuffers
[4:26:37.698151558] [4807] ERROR IPCUnixSocket ipc_unixsocket.cpp:191 Failed to send: Transport endpoint is not connected
[4:26:37.698342238] [4807] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
```
At this point I'm way above my head. Is there a way to get around this and deploy ALL of `libcamera` into my AppImage? I've already built `libcamera` from source, is there a way to build it into fewer separate binaries? Is there a way to build it into static libraries so that the linker does the job of creating a self-contained executable?

kbingham, 2023-12-10:
IPA modules are a core design feature of libcamera. This is where the image processing algorithms are contained. For some vendors this is the "secret sauce" of how they provide good image quality and they can't/don't/won't provide an open source implementation.

For this they can provide an externally loaded IPA which can not be statically linked.

From the sound of it, you don't care about that, and you are packaging all this yourself so you have full control over what goes in.

The paths are indeed over part that could be wrong, you can't just "move" libcamera. So to fix that you'd have to compile with the appropriate paths configured.

You also need to make sure the IPA signature verification is successful, or that the IPA process separation can work in your image. I would suspect making sure you have the IPA verification support at runtime in your app image is likely the main issue.

VioletGiraffe, 2023-12-10:
Thanks for replying! I understand in principle what IPA modules are, what I don't understand is why they are so difficult to load,and what I could possibly do about this. Yes, I haven't needed any vendor-specific ones, so far I only need the ones that are built from source. But never mind that. Here is the folder with the modules and the .sig files, why do they fail to load anyway?

kbingham, 2023-12-11:
Are you building libcamera from source?
I believe you haven't met the runtime dependencies required.

kbingham, 2023-12-11:
A full compile log would help identify it, or a capture with LIBCAMERA_LOG_LEVELS=*:0 set to get the full runtime debug log