kbingham/libcamera #88: Add support for OpenWrt

t3therdev, 2024-01-20:
Hi, we would love to discuss how we can bring libcamera to OpenWrt. There are many get uses for libcamera and it works perfectly using rasbian. But there are also some amazing use cases for OpenWrt and using libcamera.  

Has anyone looked at how we can support OpenWrt or cross compile, etc?

kbingham, 2024-01-20:
Hi there, happy to help. libcamera can certainly be cross compiled and packaged. I'm not aware of anyone looking at openwrt packaging yet though.

What's the package manager?

The right place to discuss this could be the libcamera mailing list.

libcamera-devel@lists.libcamera.org
https://lists.libcamera.org/listinfo/libcamera-devel

Otherwise you'll only hear from me at this repo.

kbingham, 2024-01-25:
Did you look into this any more ? I haven't seen anything on the mailing list. Cross compiling libcamera is certainly doable and done regularly. libcamera uses meson as it's build system, so you'll likely want to find another project supported by openwrt using meson as a comparison.

https://openwrt.org/docs/guide-developer/packages looks like a starting point.

I have some notes that may be useful to you when you make the package at:
 - https://github.com/kbingham/linux-cameras/wiki/Packaging-and-Distribution-of-libcamera

closed 2024-11-29 by kbingham